The-OpenROAD-Project/OpenROAD #6050: routing congestion on ihp with rtlmp enabled

gadfort, 2024-10-29:
### Describe the bug

When enabling rtlmp over the previous macro placer, the resultant design is not able to be routed.

Relates to proposed removal of previous macro placer: https://github.com/The-OpenROAD-Project/OpenROAD/pull/6047

### Expected Behavior

Design is routable

### Environment

```shell
OpenROAD v2.0-16787-gcd519bb5e 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

### To Reproduce

Placement:
[sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_place0_20241029-121603.tar.gz](https://github.com/user-attachments/files/17560020/sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_place0_20241029-121603.tar.gz)
```
tar xvf sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_place0_20241029-121603.tar.gz
cd sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_place0_20241029-121603
./run.sh
```
Routing:
[sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_route0_20241029-120252.tar.gz](https://github.com/user-attachments/files/17560031/sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_route0_20241029-120252.tar.gz)
```
tar xvf sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_route0_20241029-120252.tar.gz
cd sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_route0_20241029-120252
./run.sh
```

### Screenshots

place_macro placement:
[image]

rtlmp placement:
[image]

congresion markers:
[image]

maliberty, 2024-10-29:
@gadfort it would be helpful to have the macro placement step as well

gadfort, 2024-10-29:
First testcase:
[sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_floorplan0_20241029-131630.tar.gz](https://github.com/user-attachments/files/17560708/sc_issue_ethmac_job0_ihp130_sg13g2_stdcell_floorplan0_20241029-131630.tar.gz)

Second testcase:
[sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_floorplan0_20241029-131607.tar.gz](https://github.com/user-attachments/files/17560709/sc_issue_tiny_rocket_job0_ihp130_sg13g2_stdcell_floorplan0_20241029-131607.tar.gz)

@maliberty here you go

AcKoucher, 2025-01-13:
I looks like the main issue in both testcases wrt to macro placement is that the blockages that we generate to model pin access within MPL2 are very misleading. This is being addressed in #5669 which I'm trying to merge but problems keep arising. Using this PR's branch makes the congestion problems go away:

|  Test Case   | Placement | Route Congestion Map | RUDY
| -------- | ------- | -------- | ------- |
| ethmac  | [image] | [image] | [image] |
| tinyRocket | [image] | [image] | [image] |

PR #5809 (2025-01-22): mpl2: IOs abstraction adaptation to ORFS flow - first version (simplified)
Resolve the first top level task in #5669.
Resolve #6050.
Resolve #4867.

### Introductory Considerations
- This is the first step of the enhancements to adapt the IOs abstractions inside mpl2 to ORFS flow.
- This first version will not consider the dimensions of the constraints whether they are -exclude constraints or pins' constraint regions.

### Goals
1. Improve QoR in terms of WL.
2. Ease the life of the annealer for some cases (e.g., sky130hd/uW) that require extreme virtual std cell utilization configurations to  [... cut]
closed 2025-01-22 by eder-matheus