The-OpenROAD-Project/OpenROAD #6055: macro placement fails

oharboe, 2024-10-29:
### Describe the bug

with SYNTH_HIERARCHICAL=0, number of instances is half of that of SYNTH_HIERARCHICAL=1, so it was worth trying to turn it off, but macro placement fails. https://github.com/The-OpenROAD-Project/megaboom/pull/188

untar https://drive.google.com/file/d/1wvt-ThCrWKvSeo4jKmxmgpnbLtjV_XRE/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-1.sh 
OpenROAD v2.0-16664-g9ba039016 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
rtl_macro_placer -halo_width 19 -halo_height 19 -report_directory .//objects/asap7/BoomTile/1/rtlmp -target_util 0.24
Floorplan Outline: (0.0, 0.0) (2000.0, 2000.0),  Core Outline: (2.052, 2.16) (1998.0, 1998.0)
	Number of std cell instances: 1046614
	Area of std cell instances: 127052.54
	Number of macros: 72
	Area of macros: 810855.88
	Halo width: 19.00
	Halo height: 19.00
	Area of macros with halos: 1394272.12
	Area of std cell instances + Area of macros: 937908.44
	Core area: 3983592.75
	Design Utilization: 0.24
	Core Utilization: 0.04
	Manufacturing Grid: 1

[ERROR MPL-0040] Failed on cluster (root)_glue_logic_1_0_0
Error: macro_place_util.tcl, 76 MPL-0040
Command exited with non-zero status 1
Elapsed time: 39:50.24[h:]min:sec. CPU time: user 6409.13 sys 687.58 (296%). Peak memory: 16733712KB.
```

### Expected Behavior

Successful macro placement

### Environment

```shell
OpenROAD v2.0-16664-g9ba039016
```

### To Reproduce

See above.

oharboe, 2024-10-30:
Note that if I change the PLACE_DENSITY slightly, then macro placement passes. I tried with 0.22, 0.30, 0.20, 0.25, 0.22: macro placement works in all cases.

maliberty, 2024-10-30:
Such is the nature of annealing.

AcKoucher, 2024-10-30:
@oharboe FYI I'm investigating what is going on for #5996 and I think that it might be same problem across multiple issues, including this one.

PR #6090 (2024-11-05): mpl2: ensure that sa converges if a valid result was generated at any point
Resolves #5996,
Resolves #6031,
Resolves #6055.

When I implemented #5908 I missed one fundamental thing: the cluster placement result (and all the other SoftMacro annealing results) is not just the sequence pair itself but rather a combination of the sequence pair along with the specific shapes chosen during a perturbation iteration. _I.e., the same sequence pair can represent different floorplan arrangements depending on the clusters' shapes._

As this only applies for SoftMacro annealing, even though the logic for s [... cut]
closed 2024-11-05 by maliberty