The-OpenROAD-Project/OpenROAD #6063: initialize_floorplan fails without -site option

titan73, 2024-10-30:
### Describe the bug

The - site is supposed to be optional. A warning is printed but an error occurs later.

% initialize_floorplan -utilization 10
[WARNING IFP-0011] use -site to add placement rows.
TypeError in method 'init_floorplan_util', argument 7 of type 'odb::dbSite *'

% initialize_floorplan -die_area {0 0 10 10} -core_area {0 0 10 10}
[WARNING IFP-0011] use -site to add placement rows.
TypeError in method 'init_floorplan_core', argument 9 of type 'odb::dbSite *'

Is -site option really to be optional? Seems the contrary. In that case the warning should be turn into an error and doc updated.

### Expected Behavior

No error.

### Environment

```shell
Lastest git.
```

### To Reproduce

Type:
    initialize_floorplan -utilization 10
or
    initialize_floorplan -die_area {0 0 10 10} -core_area {0 0 10 10}

maliberty, 2024-10-30:
It looks like it should be an error.  Do you want to make a PR?

titan73, 2024-10-30:
Yes. I'll do it. Assign this issue to me.

titan73, 2024-10-30:
Sent PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/6069

titan73, 2024-12-11:
The PR has been merged. Closing.