The-OpenROAD-Project/OpenROAD #6073: GRT: store route guides even when it fails with congestion

eder-matheus, 2024-10-31:
### Description

Today we don't have the option to see route guides of nets when the design has congestion. This makes debug harder, since having only the markers with the congestion regions don't say much about the routing each net is taking.

### Suggested Solution

Complete the global route flow even when routing congestion is detected, and store the guides in odb.
The detailed router needs to be updated too, in order to avoid routing a congested design.

gadfort, 2024-10-31:
Wouldn't it be better to allow GRT to mark the guides as "dirty" or "unusable" that way DRT only needs to check a simple flag instead of needing to check for congestion (since we have flags that allow for GRT to finish with congestion and DRT is capable of handling a little bit of congestion)

eder-matheus, 2024-10-31:
> Wouldn't it be better to allow GRT to mark the guides as "dirty" or "unusable" that way DRT only needs to check a simple flag instead of needing to check for congestion (since we have flags that allow for GRT to finish with congestion and DRT is capable of handling a little bit of congestion)

I'm planning to use the congestion markers stored in odb to detect that. In the final implementation, DRT would need only a single function call to check that.

closed 2024-11-06 by eder-matheus