The-OpenROAD-Project/OpenROAD #6079: Narrowing layout view turns blockages into checkerboard

jeffng-or, 2024-10-31:
### Describe the bug

Minor issue but filing for tracking.

When viewing macro placement blockages in the GUI, the stippling causes the blockage shapes to become a checkerboard-like pattern that doesn't surround the macro:

[image]

If you continue to narrow the widget, it cycles through a couple of different visual representations before becoming checkerboard again:

A little narrower (this looks as expected):
[image]

A little narrower still (starting to get checkerboardy but shapes still surround macro):

[image]

I uploaded a video: https://drive.google.com/file/d/1m45glTVljXfBoTJrAEWOHoXTcXholIo7/view?usp=sharing

Note that when you zoom in, the blockages are drawn correctly.

### Expected Behavior

Some other indication that the blockage is there

### Environment

```shell
OpenROAD v2.0-16787-gcd519bb5e
```

### To Reproduce

untar https://drive.google.com/file/d/1vBCqT6eZNc4hQky2EKP3htb7aXHH1CYt/view?usp=sharing

Run ./run-me-ICache-asap7-base.sh gui to bring up GUI
Start narrowing the layout widget

maliberty, 2025-05-23:
@gadfort interested?

gadfort, 2025-05-23:
@maliberty I don't have time right now, but I suspect it has to do with some aliasing (when I looked at it before, there are references to this behavior on Qts issues as well.)