The-OpenROAD-Project/OpenROAD #6089: segfault in report_checks

gadfort, 2024-11-04:
### Describe the bug

Relates to but is a different segfault: https://github.com/The-OpenROAD-Project/OpenROAD/issues/6025

Running report_checks -> pdngen -> remove_buffers -> report_checks causes a segfault

It's related to setting the signal type in PDN for the BTerm (if I comment that out the segfault goes away).

### Expected Behavior

No segfault and just works

### Environment

```shell
head of master
```

### To Reproduce

https://drive.google.com/file/d/1Z_l81zkoTRBQQXe19CcY90h5oMKdolQu/view?usp=sharing

```
tar xvf sc_issue_zerosoc_core_job0_floorplan0_20241103-190408.tar.gz
cd sc_issue_zerosoc_core_job0_floorplan0_20241103-190408
./run.sh
```

### Relevant log output

```shell
[INFO RSZ-0026] Removed 913 buffers.
SC_METRIC: report_checks -path_delay max
/usr/include/c++/13/bits/stl_vector.h:1147: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](size_type) const [with _Tp = sta::TableBlock<sta::Vertex>*; _Alloc = std::allocator<sta::TableBlock<sta::Vertex>*>; const_reference = sta::TableBlock<sta::Vertex>* const&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x000060CFB67E1C63 in /home/pgadfort/.local/bin/openroad
 1# 0x00007DD3C6445320 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007DD3C68DF90D in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x000060CFB6AC13F7 in /home/pgadfort/.local/bin/openroad
 7# sta::Sim::annotateVertexEdges(sta::Instance const*, bool) in /home/pgadfort/.local/bin/openroad
 8# sta::Sim::ensureConstantsPropagated() in /home/pgadfort/.local/bin/openroad
 9# sta::Sta::searchPreamble() in /home/pgadfort/.local/bin/openroad
10# sta::Sta::findPathEnds(sta::ExceptionFrom*, sta::Vector<sta::ExceptionThru*>*, sta::ExceptionTo*, bool, sta::Corner const*, sta::MinMaxAll const*, int, int, bool, float, float, bool, sta::Set<char const*, sta::CharPtrLess>*, bool, bool, bool, bool, bool, bool) in /home/pgadfort/.local/bin/openroad
11# 0x000060CFB6A5F53D in /home/pgadfort/.local/bin/openroad
12# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
13# 0x00007DD3CAFB66BC in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
16# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/pgadfort/.local/bin/openroad
17# 0x000060CFB67E1446 in /home/pgadfort/.local/bin/openroad
18# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
19# main in /home/pgadfort/.local/bin/openroad
20# 0x00007DD3C642A1CA in /lib/x86_64-linux-gnu/libc.so.6
21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
22# _start in /home/pgadfort/.local/bin/openroad
```

gadfort, 2024-11-04:
@maliberty indicates that the networkChanged callback isn't clearing enough

maliberty, 2024-11-04:
@jjcherry56 this appears to be an sta issue.  

From what I can tell when the vdd pin is reached, and it is a constant, then Sim::annotateVertexEdges does `Vertex *vertex = graph_->pinDrvrVertex(pin);` which leads to an invalid access in the vertices_ table as there is no vertex for this pin.

parallaxsw, 2024-11-05:
The bterm has a stale "pointer" to a vertex so it needs to be nulled.
This isn't ideal but but it works for now.
[issue6089.patch](https://github.com/user-attachments/files/17635236/issue6089.patch)

PR #6095 (2024-11-05): dbSta: workaround for stale vertex it on sig type change
Fixes #6089
commit 57e6dc0b (2024-11-05): dbSta: workaround for stale vertex it on sig type change

Fixes #6089
closed 2024-11-05 by maliberty