The-OpenROAD-Project/OpenROAD #6097: DRC violations due to routing to strict about following tracks when accessing pins

titan73, 2024-11-05:
### Describe the bug

After 18 iterations, there are 40 similar drc violations still not fixed. 

M2 route follow track when switching to M1 causing DRC. 

[image]

Pin access in Innovus

[image]

### Expected Behavior

M2 should not strictly follow track if possible to fix DRC.

### Environment

```shell
v2.0-16855-ga15114a95
```

### To Reproduce

openroad test.tcl with test case below:

[drc_pin_access.zip](https://github.com/user-attachments/files/17636330/drc_pin_access.zip)

maliberty, 2024-11-05:
Please don't include proprietary tool output.

stefanottili, 2024-11-05:
On the picture it looks as if your rows are not aligned with the M1/M2 tracks.
I that leads to the stdcell pins not being aligned either and should be the root cause of pin access issues.

osamahammad21, 2024-11-05:
@stefanottili I agree with you, but I don't understand why pin access did not complain about this earlier in the flow. I'll investigate further.

titan73, 2024-11-06:
> In the picture it looks as if your stdcell sites are not a multiple of the routing pitch. This is very unusual and leads to none of the stdcell pins being "on track".

Yes indeed, the sites for this library is not multiple of the pitch. This library is badly designed. All other sites for other libraries in this techno are correct.

maliberty, 2024-11-06:
Do you need to use the badly designed library?  I'm not saying we shoudn't look at it but it also seems like its making things unnecessarily hard.

titan73, 2024-11-07:
Yes. There is not replacement for this standard cells library (215 cells). It's a mature techno with a lot of products based on it. They unfortunately won't redesign these cells.

titan73, 2024-11-07:
Btw, apart from the site width not multiple of pitch, if a library has the correct site but the cells pins are not or badly aligned, that will cause the same issues in OR, right?

stefanottili, 2024-11-07:
Why don’t you use a pitch that matches the design of the stdcells ?

titan73, 2024-11-07:
> Why don’t you use a pitch that matches the design of the stdcells ?

I plan to test this with custom tracks. This could work in that case because there is only one power domain using GO1 cells. With 2 power domains that won't because GO2 cells have a correct sites so if we change the tracks for GO1's ones cells this will break the GO2's ones.

osamahammad21, 2024-11-10:
@bnmfw I'm concerned that pin access allows access to this pin while it's impossible with the given access points and the nearby cell. Something is probably going wrong in pa pattern step for generating access patterns for a row of cells.

maliberty, 2024-11-11:
While we look at this, a simple workaround is to use
```
global_placement -pad_left 1 -density 0.8
set_placement_padding -global -left 1
```

maliberty, 2024-11-11:
It looks like the row based search finds the lowest cost set of APs but they are not DRC free.  I suppose it hopes that dr will find a solution and accepts it.  Really we need to generate at least one AP that is not so close to the edge.

maliberty, 2024-11-11:
`-min_access_points 5` also fixes the problem.

maliberty, 2024-11-11:
@bnmfw I think we should the termination at MINNUMACCESSPOINT_STDCELLPIN to also consider if all APs are so close to the cell edge that such a failure could occur when we do row testing.  We should try to find at least one not so close before stopping even if we hit the overall minimum.

bnmfw, 2025-04-01:
@maliberty @osamahammad21 In this case, what is a point "not so close" do the instance edge? In #6889 I experimented with some distances, and found that twice the layer width (`design_->getTech()->getLayer(ap->getLayerNum())->getWidth() * 2`) gave the best results for this specific case. The overall DRT flow still ended with DRVs but it reached the minimum of 4 during incremental repair instead of 55. I'm not sure if this value makes any sense, I just tested multiples of the layer width (0, 2/3, 1, 2, 3, 4) and empirically found 2 to be the best.

However this new soft restriction introduces a single antenna error on two public CI tests. Matt said that was OK so I was wondering if there could be a case for a merge. I did not even open the PR because of these CI errors.

maliberty, 2025-04-01:
The most thorough solution would be to drive the "not so close" with the DRC rules assuming a shape at the minimum possible distance outside the cell.  I don't want to spend too much effort on this corner case and think just picking 2 is fine so long as the runtime increase in "normal" designs isn't too large.

I think there is a case for doing a PR and moving on.

osamahammad21, 2025-04-24:
@bnmfw any reason why this is not caught in prepPatternInstRows? Is it seen accessible by pa but not by dr?

PR #6889 (2025-04-02): drt: pa far from edge
Fixes #6097. Almost.
Secure and ISPD CI all passed, the only violations are on regression tests. Ibex_sky130hd has 1 metric failure:
```
Pin                                     Limit      Cap    Slack
---------------------------------------------------------------
_23420_/Y                               0.092    0.094   -0.001 (VIOLATED)
```
and jpeg_sky130hd as well as aes_sky130hd have 1 antenna violation each.

This PR introduces the soft requirement that a pin has an Access Point at least 2 times the layer width away from the instance edge. This seems to help the issu [... cut]