The-OpenROAD-Project/OpenROAD #6111: segfault when calling place_pin without tracks

gadfort, 2024-11-07:
### Describe the bug

When calling `place_pin` with `-force_to_die_boundary` in a design without tracks defined openroad segfaults with
```
Signal 11 received
Stack trace:
 0# 0x000059DDF3B26A63 in openroad
 1# 0x00007BC341C45320 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbTrackGrid::getGridPatternY(int, int&, int&, int&) in openroad
 3# ppl::IOPlacer::movePinToTrack(odb::Point&, int, int, int, odb::Rect const&) in openroad
 4# ppl::IOPlacer::placePin(odb::dbBTerm*, odb::dbTechLayer*, int, int, int, int, bool, bool) in openroad
 5# 0x000059DDF3CE5E7A in openroad
 6# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007BC34678B6BC in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
10# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in openroad
11# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) in openroad
12# 0x000059DDF3B25D6E in openroad
13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# main in openroad
15# 0x00007BC341C2A1CA in /lib/x86_64-linux-gnu/libc.so.6
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in openroad
Segmentation fault (core dumped)
```

### Expected Behavior

Generate a error indicating that tracks are missing

### Environment

```shell
v2.0-16896-g1531ad983
```

### To Reproduce

[seg-fault.tar.gz](https://github.com/user-attachments/files/17663427/seg-fault.tar.gz)
```
tar xvf seg-fault.tar.gz
cd off-track
openroad run.tcl
```

### Relevant log output

```shell
Signal 11 received
Stack trace:
 0# 0x000059DDF3B26A63 in openroad
 1# 0x00007BC341C45320 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbTrackGrid::getGridPatternY(int, int&, int&, int&) in openroad
 3# ppl::IOPlacer::movePinToTrack(odb::Point&, int, int, int, odb::Rect const&) in openroad
 4# ppl::IOPlacer::placePin(odb::dbBTerm*, odb::dbTechLayer*, int, int, int, int, bool, bool) in openroad
 5# 0x000059DDF3CE5E7A in openroad
 6# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007BC34678B6BC in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
10# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in openroad
11# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) in openroad
12# 0x000059DDF3B25D6E in openroad
13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# main in openroad
15# 0x00007BC341C2A1CA in /lib/x86_64-linux-gnu/libc.so.6
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in openroad
Segmentation fault (core dumped)
```

maliberty, 2024-11-07:
@gadfort you will get an error instead - is that expected?

maliberty, 2024-11-07:
If no wrong way routing is allowed then that is reasonable however OR would never generate such.  Is that what you are seeing here?  I suspect problems will occur elsewhere too.  @gadfort is that intended?

gadfort, 2024-11-07:
@maliberty to be honest I only stumbled into this issue while generating another testcase, so I don't know if the pins at that time are correct.
I think is rect-only and non-preferred direction (error), if non-preferred (I think it should be acceptable with a warning maybe). If no tracks available, just error

closed 2024-11-07 by eder-matheus