The-OpenROAD-Project/OpenROAD #6112: drt ends with DRCs when accessing pins that are slightly off grid on rect only layer

gadfort, 2024-11-07:
### Describe the bug

When routing to macro pins that are not aligned with the routing grid on rect only layers, DRT fails to route cleanly even though there is plenty of available resources and it would only require small stub on M6 that would be offtrack to route cleanly.

### Expected Behavior

DRT routes cleanly

### Environment

```shell
OpenROAD v2.0-16896-g1531ad983 
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

[off-track-route.tar.gz](https://github.com/user-attachments/files/17663578/off-track-route.tar.gz)

```
tar xvf off-track-route.tar.gz
cd off-track
openroad run.tcl
```

### Relevant log output

```shell
[INFO DRT-0195] Start 64th optimization iteration.
    Completing 10% with 7 violations.
    elapsed time = 00:00:00, memory = 6237.39 (MB).
    Completing 20% with 7 violations.
    elapsed time = 00:00:00, memory = 6237.39 (MB).
    Completing 30% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 40% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 50% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 60% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 70% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 80% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 90% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
    Completing 100% with 7 violations.
    elapsed time = 00:00:01, memory = 6237.39 (MB).
[INFO DRT-0199]   Number of violations = 7.
Viol/Layer          M6     V6     M7
Metal Spacing        1      0      0
Rect Only            2      0      0
Short                2      1      1
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 6237.39 (MB), peak = 6417.02 (MB)
```

### Screenshots

M5-M6: 
[image]

M6-M7:
[image]

### Additional Context

@maliberty this is a small testcase replicating the issue I was describing the other day.

osamahammad21, 2024-11-10:
M6 and M7 have LEF58_RIGHTWAYONGRIDONLY rule which prevents DRT from routing offtrack. What is your suggested solution?

gadfort, 2024-11-10:
@osamahammad21 I didn't realize it had that rule in there. In that case, I would say you should error out during pin access, there is no way to access that pin so no need to spend time routing this.

maliberty, 2024-11-11:
I think I have a fix for this.  Running a secure CI.  FlexPA::genAPsFromRect should not set use_center_line if this constraint is present.

closed 2024-11-11 by maliberty

maliberty, 2024-11-11:
I get:
[image]

gadfort, 2024-11-11:
@maliberty you are correct, I was still running against the previous version (user error). Thanks

PR #6132 (2024-11-11): drt: check for on-grid-only during PA of macro pins
Fixes #6112
commit 63f1e554 (2024-11-11): drt: check for on-grid-only during PA of macro pins

Fixes #6112