The-OpenROAD-Project/OpenROAD #6113: make_tracks on asap7 generates an error `x_offset -270 not >= 0`

gadfort, 2024-11-07:
### Describe the bug

running `make_tracks` on asap7 generates `[ERROR IFP-0039] x_offset -270 not >= 0`

### Expected Behavior

Either improve the error message to use correct units and indicate the offending metal layer or correctly generate the routing tracks.

### Environment

```shell
OpenROAD v2.0-16896-g1531ad983 
Features included (+) or not (-): +Charts +GPU +GUI +Python
```

### To Reproduce

[make_tracks.tar.gz](https://github.com/user-attachments/files/17663788/make_tracks.tar.gz)
```
tar xvf make_tracks.tar.gz
cd off-track
openroad run.tcl
```

### Relevant log output

```shell
[INFO IFP-0001] Added 5999 rows of 29999 site asap7sc7p5t.
[ERROR IFP-0039] x_offset -270 not >= 0
Error: run.tcl, 9 IFP-0039
```

maliberty, 2024-11-07:
> correctly generate the routing tracks.

I don't think we want tracks outside the die area

gadfort, 2024-11-07:
@maliberty I think you could create the tracks, but skip the ones that would fall outside the diearea. I don't feel strongly about it, but I would prefer to error message to be in um and reference the metal layer.

PR #6155 (2024-11-14): Improved IFP error messages
Closes #6113.

Changes:
- Improved error messages. Units converted to micrometer (μm).
- Duplicate error IDs removed.
closed 2024-11-14 by maliberty