The-OpenROAD-Project/OpenROAD #6120: odb: write_odb does not work work anymore

titan73, 2024-11-07:
### Describe the bug

See title.

### Expected Behavior

write_db should succeed.

### Environment

```shell
v2.0-16855-ga15114a95
```

### To Reproduce

openroad test.tcl

[write_db.zip](https://github.com/user-attachments/files/17669683/write_db.zip)

eder-matheus, 2024-11-07:
@titan73 If I manually create the dbs directory, the command don't fail. I'm not sure if any of our write commands should create directories. @maliberty any comments?

titan73, 2024-11-07:
@eder-matheus yeah indeed I've just figured out that dbs folder was removed but the error message was not very helpful and I couldn't pinpoint the problem. I don't expect write_db to create the folder that that would be good to have a error message complaining about the missing "dbs" folder.

maliberty, 2024-11-07:
What was the message?

maliberty, 2024-11-07:
I don't think there is any change here so "anymore" seems inaccurate.

eder-matheus, 2024-11-07:
> What was the message?

@maliberty This: `Error: test.tcl, 8 basic_ios::clear: iostream error (failed to open 'dbs/floorplan.odb.1'): iostream error`.

My guess is that we are not checking if the file was properly created before trying to write to it.

maliberty, 2024-11-08:
"failed to open" is pretty clear that it is a file system error but we could make it prettier.

titan73, 2024-11-08:
Updated title.

Indeed the message says what's wrong. I got confused with its complicated message (like an assert) and thought there was a bug.
So nothing wrong actually but yes a better message could be nice.
Btw is this message sent to the logger?
And I don't understand. The message refers to file 'dbs/floorplan.odb.1' but the file is 'dbs/floorplan.odb'. What is this '.1'?

titan73, 2024-11-11:
Fixed  by merged PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/6133.

closed 2024-11-11 by titan73