The-OpenROAD-Project/OpenROAD #6149: gui: drc violation blinking disappears after selecting violation in drc viewer

titan73, 2024-11-13:
### Describe the bug

In drc viewer, mousing over a violation make it highlighted with blinking yellow in the renderer but when clicking on this very violation the highlighting disappears. That would be more natural to keep it highlighted.

In addition to this issue a possible enhancement could be to keep the selected violation highlighted when mouse cursor is not on any violations.

### Expected Behavior

Highlight is kept when clicking on a violation.

### Environment

```shell
v2.0-16941-g1452788e1
```

### To Reproduce

Whatever design with drc violations.

titan73, 2024-11-13:
You can assigned it to me. I have make it working and will send a PR.

gadfort, 2024-11-13:
@titan73 do you have `highlight selected` turned off?

titan73, 2024-11-13:
Yes it was was off. I turned it on. The selection becomes yellow but there is not blinking like we have with the mouse over.

titan73, 2024-12-12:
@gadfort Should we still keep this issue opened?
As you said in related PR:
`It seems like this exposes and masks some odd behavior of the GUI, but in the short run it seems fine. I think we should keep the issue open to track fixing the inspector animation behavior.`
Shouldn't we close this issue and open a new one for other issues?

maliberty, 2024-12-15:
What was the related PR?  I see nothing to fix here.  If there is another issue please open a new one.

closed 2024-12-15 by maliberty

titan73, 2024-12-15:
> What was the related PR? I see nothing to fix here. If there is another issue please open a new one.

The PR is https://github.com/The-OpenROAD-Project/OpenROAD/pull/6150 which has been merged.