The-OpenROAD-Project/OpenROAD #6152: [IHP sg13g2] IO pin access problem on the chip right side only

lanserge, 2024-11-13:
### Describe the bug

Global routing failed reporting congestion at the right side IO cells only. However there is no real congestion.
With some experimentation I could say that removing OBS from IO cells in LEF results in successful routing, but those OBS are actually on opposite side of IO cell and are not obstacles for routing. And actually other sides of the chip have same IO and routed without issue. Looks like some issue related to the cells orientation in the OpenROAD with determining pin access.

### Expected Behavior

All sides of the chip routed.

### Environment

```shell
kernel: Linux 6.10.11+bpo-amd64
os: Debian GNU/Linux 12 (bookworm)
cmake version 3.30.5
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-16972-g32d202458
-- System name: Linux
-- Compiler: GNU 12.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
[... 86 lines cut ...]
-- Number of processor cores: 4
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")
-- Found Eigen3: /home/serge/projects/chipflow-backend-tools/tools/or-tools/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (5.8s)
-- Generating done (0.9s)
```

### To Reproduce

[5_1_grt.zip](https://github.com/user-attachments/files/17736281/5_1_grt.zip)
Unzip attached archive (it contains all the dependencies), from inside 5_1_grt folder run make.
The global routing will fail, produced files for analysis will be `5_1_grt/results/5_1_grt-failed.odb` and `5_1_grt/reports/congestion.rpt`

### Screenshots

[image]

lanserge, 2024-11-20:
May be this will be of help: it some kind of related to Metal2 layer, as removing all the OBS only on this layer from IO LEF results in success. The questions are still the same: why chip right side only and why this cause congestion as those Metal2 OBS rectangles are on opposite side of pin routing.

closed 2025-01-07 by eder-matheus

eder-matheus, 2025-01-07:
@lanserge Fixed, please checkout to the latest master branch to get the fix. It was a pin access issue in the end. Let me know if you face other issues.