The-OpenROAD-Project/OpenROAD #6172: `make gui_grt` with latest ORFS no longer opens the GUI

oharboe, 2024-11-17:
### Describe the bug

Check out latest ORFS, run `./build_openroad.sh --local --latest`, then use nangate/gcd/config.mk:

```
$ make clean_all grt
$ make gui_grt
ODB_FILE=./results/nangate45/gcd/base/5_1_grt.odb GUI_SHOW=1 /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  -minimize /home/oyvind/OpenROAD-flow-scripts/flow/scripts/open.tcl
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-17042-g8a24d74d8 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
GUI_TIMING=1 reading timing, takes a little while for large designs...
estimate_parasitics -global_routing
Populating timing paths...
[nothing more happens, the GUI is not opened]
```

I can see that OpenROAD is running and using 0% CPU, if I click on the icon, it opens normally:

[image]

### Expected Behavior

`make gui_grt` should open the GUI after timing has been loaded

### Environment

```shell
Git commit: 8a24d74d8f961a4a1e12b985162b2b12f7973b9d
kernel: Linux 6.8.0-48-generic
os: Ubuntu 24.04.1 LTS (Noble Numbat)
cmake version 3.28.3
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17042-g8a24d74d8
-- System name: Linux
-- Compiler: GNU 13.2.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- PerformingCMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR) (Required is at
  least version "4.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.28/Modules/FindSWIG.cmake:153 (find_package_handle_standard_args)
  src/CMakeLists.txt:169 (find_package)

 Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17 - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17 - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Configuring incomplete, errors occurred!
```

### To Reproduce

See above

oharboe, 2024-11-17:
Tried `make OR_ARGS=-no_init gui_grt`, no difference.

oharboe, 2024-11-17:
Tried `QT_QPA_PLATFORM=xcb make OR_ARGS=-no_init gui_grt`, no difference.

maliberty, 2024-11-17:
`gui::minimize` work when the gui is showing?

gadfort, 2024-11-17:
@maliberty I get the same behavior on Ubuntu24.
Ie. `showNormal` doesn't seem to do anything.

If I run with:
```
gui::minimize
gui::unminimize
```

Nothing happens to open the window again.

oharboe, 2024-11-17:
> It opens for me.  On what OS?  Under docker?

No docker, Ubuntu 24.04 (as per logs).

maliberty, 2024-11-17:
Does showMaximized() work?  Does show() help?

maliberty, 2024-11-17:
Or `w->setWindowState((w->windowState() & ~Qt::WindowMinimized) | Qt::WindowActive);` (https://doc.qt.io/qt-5/qwidget.html#setWindowState)

gadfort, 2024-11-17:
@maliberty short answer is Qt6 gets rid of some warnings I see related to Wayland. If I disable Wayland temporarily it seems to work, but this appears to be an issue in Qt.

maliberty, 2024-11-17:
@gadfort does qt6 solve the issue?

gadfort, 2024-11-18:
@maliberty no it did not.

oharboe, 2024-11-21:
> @maliberty short answer is Qt6 gets rid of some warnings I see related to Wayland. If I disable Wayland temporarily it seems to work, but this appears to be an issue in Qt.

How do I disable Wayland in Qt?

oharboe, 2024-11-22:
@maliberty If this will take a while, can we add a workaround to ORFS meanwhile?

oharboe, 2024-11-30:
Possible workaround: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2600

oharboe, 2024-11-30:
Hmm... that workaround works on designs/nangate45/gcd/config.mk gui_floorplan, but when I try it on megaboom `make gui_grt`, the GUI doesn't open up, but the taskbar icon jiggles:

[Screencast from 2024-11-30 15-54-35.webm](https://github.com/user-attachments/assets/da936a5a-e2d2-4ce8-b029-36e358296fab)

oharboe, 2024-11-30:
Throwing stuff at the wall to see if something sticks. This does not fix it:

```c++
$ git diff
diff --git a/src/gui/src/gui.cpp b/src/gui/src/gui.cpp
index 53b36c019..4205836b2 100644
--- a/src/gui/src/gui.cpp
+++ b/src/gui/src/gui.cpp
@@ -1273,6 +1273,14 @@ void Gui::minimize()
 void Gui::unminimize()
 {
   main_window->showNormal();
+  if (main_window->isMinimized()) {
+    main_window->setWindowState(
+        (main_window->windowState() & ~Qt::WindowMinimized) | Qt::WindowActive);
+  }
+  main_window->showNormal();
+  main_window->raise();
+  main_window->activateWindow();
+  main_window->setFocus(Qt::OtherFocusReason);
 }
```

oharboe, 2024-11-30:
More random googled stuff that doesn't work:

```C++
+#include <X11/Xlib.h>
+#include <X11/Xutil.h>
+
 void Gui::unminimize()
 {
-  main_window->showNormal();
+  Display *display = XOpenDisplay(nullptr);
+Window window = main_window->winId();
+
+XEvent xev;
+memset(&xev, 0, sizeof(xev));
+xev.xclient.type = ClientMessage;
+xev.xclient.message_type = XInternAtom(display, "_NET_ACTIVE_WINDOW", False);
+xev.xclient.format = 32;
+xev.xclient.window = window;
+xev.xclient.data.l[0] = 1;  // source indication
+xev.xclient.data.l[1] = CurrentTime;
+
+XSendEvent(display, DefaultRootWindow(display), False,
+           SubstructureNotifyMask | SubstructureRedirectMask, &xev);
+XFlush(display);
```

maliberty, 2024-12-18:
I spun up an ubuntu24 machine on gcp and ran `openroad -no_init -gui -minimize x` where x contains `gui::unminimize` and the gui appeared with no problem.

I still can't reproduce your issue.  @gadfort can you?

maliberty, 2024-12-18:
`make gui_final` on ubuntu24 also worked fine.

gadfort, 2024-12-19:
@maliberty it still does not work for me.  Are you using wayland?
```
>>> echo $XDG_SESSION_TYPE 
wayland
```

maliberty, 2024-12-19:
I get `tty` but if I set it to wayland it makes no difference.  How do I get wayland to happen?

maliberty, 2024-12-19:
Do I need to have an nvidia card to get wayland?  I spun up a more basic machine.

maliberty, 2024-12-19:
No difference with a GPU.  Still no idea how to reproduce it.

gadfort, 2024-12-19:
You don't need an Nvidia card, my machine just has integrated AMD graphics. I'm not familiar with the gcp machines but the azure and AWS machines have some odd default settings that differ from a machine you would just install locally.

oharboe, 2024-12-20:
I have tried to articulate a Qt standalone issue https://bugreports.qt.io/browse/QTBUG-132390 :crossed_fingers:

closed 2024-12-26 by maliberty