The-OpenROAD-Project/OpenROAD #6173: Add OpenROAD icon to Ubuntu task bar

oharboe, 2024-11-17:
### Description

When OpenROAD is minimized, I get a generic icon, would be nice with an OpenROAD icon to easily spot OpenROAD:

[image]

### Suggested Solution

Add icon

gadfort, 2024-11-17:
@maliberty I can take a look at this since I've noticed this as well.

maliberty, 2024-11-17:
I'm using 22.04 LTS fwiw

gadfort, 2024-11-17:
@maliberty I'm on 24.04, which seems to rely on a slightly different system.

oharboe, 2024-11-18:
Still no icon:

[image]

gadfort, 2024-11-18:
@oharboe if OpenROAD is installed in a non-standard location, the desktop file will not be in the correct place and will need to copied to the correct place. I suspect this is an issue with your installation

oharboe, 2024-11-18:
> @oharboe if OpenROAD is installed in a non-standard location, the desktop file will not be in the correct place and will need to copied to the correct place. I suspect this is an issue with your installation. If you are installing using orfs, then it's an orfs issue

My usecase is to build and run OpenROAD without installing it into the OS.

Can icons work with relative paths?

gadfort, 2024-11-18:
@oharboe no, the .desktop file must appear in a particular set of directories (as per freedesktop). This is a limitation of the Ubuntu desktop system. You will need the ORFS installer to copy that file to the correct place (`$home/.local/share/applications`) is sufficient. The location of the icon itself doesn't matter.

oharboe, 2024-11-18:
I see. In that case, I dont see an icon as possible/practical for my usecase.