The-OpenROAD-Project/OpenROAD #6175: Slight difference in results when using -corners

maliberty, 2024-11-17:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6174

<div type='discussions-op-text'>

<sup>Originally posted by **kareefardi** November 17, 2024</sup>
[test.tar.gz](https://github.com/user-attachments/files/17791107/test.tar.gz)
This is a test case of two scripts running global placement. Both functionally should be identical. The difference is one uses `-corner` for `set_wire_rc` while the other doesn't.

For the case using `-corner`, `set_layer_rc` is needed. The code attempts to fetch the default values in the tlef and reuse them in `set_layer_rc` after doing the necessary unit conversions. 

There seems to be a slight difference in the output generated from both runs (notice the reported slacks and HPWL) and I am wondering if this a round-off error, a methodology error at my end or something else related to OR itself. </div>

kareefardi, 2025-01-02:
Hello. Any progress on this issue?

gudeh, 2025-01-02:
Hi @kareefardi, I see the difference between runs only happens after the first timing-driven iteration. I am not yet fully aware of the implications of the set_wire_rc command from rsz.

Are you sure the two runs should be equivalent?
[image]

kareefardi, 2025-01-05:
> Hi @kareefardi, I see the difference between runs only happens after the first timing-driven iteration. I am not yet fully aware of the implications of the set_wire_rc command from rsz.
>
> Are you [...]

Hello @gudeh. From a usage point of view, I believe they should be equivalent.
I am trying to understand where is the difference coming from and if I am incorrectly using the APIs - or perhaps there is round off differences somewhere.

kareefardi, 2025-01-05:
I guess this can also be avoid by understanding this error:

```
[INFO GPL-0030] NumBins: 4096
[WARNING RSZ-0088] Corner: nom_tt_025C_1v80 has no wire signal resistance value.
[ERROR RSZ-0089] Could not find a resistance value for any corner. Cannot evaluate max wire length for buffer. Check over your `set_wire_rc` configuration
```

This happens when `set_wire_rc` is called with the argument `-corner`. It is not clear what is the impact of not using the argument when there are multiple (or single) corner defined

maliberty, 2025-01-05:
FYI, test.tar.gz is misnamed as it is not compressed and should be test.tar

kareefardi, 2025-01-05:
> FYI, test.tar.gz is misnamed as it is not compressed and should be test.tar

Thanks

arthurjolo, 2025-01-06:
> Could not find a resistance value for any corner

> I guess this can also be avoid by understanding this error:
>
> ```
> [INFO GPL-0030] NumBins: 4096
> [WARNING RSZ-0088] Corner: nom_tt_025C_1v80 has no wire signal resistance value.
> [ERROR RSZ-0089] Could [...]

Also I was not able to reproduce this warnings.

kareefardi, 2025-01-08:
> Hi @kareefardi I did some debugging and I believe there are some floating point errors here.
> 
> I changed your tcl script to show the values of RC per layer, the original value and the value after you use the command `set_layer_rc`. ![image](https://private-user-images.githubusercontent.com/78425256/400500972-cd1f0cfa-ec39-4762-a92e-a585dd45cf82.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzMjUxMzMsIm5iZiI6MTczNjMyNDgzMywicGF0aCI6Ii83ODQyNTI1Ni80MDA1MDA5NzItY2QxZjBjZmEtZWMzOS00NzYyLWE5MmUtYTU4NWRkNDVjZjgyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA4VDA4MjcxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlNjU4NGMzN2Q0ZjhmN2ZjNTUxYjk3MzNlZjhiYmIyMmE0NmU1NTI1MDc2NmE4Njc0NzRiYjdlNWFiZjZlZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oCkgsRC6gkkFwKsqlMN21WspHjWU-Q6lHzGx88UwT4c)
> 
> On the image I printed the values for layer met2. On the top is the value read from `rsz::dblayer_wire_rc`, and on the bottom is the value read from `rsz::layer_resistance` and `rsz::layer_capacitance` after you called `set_layer_rc` (the command that is used on `set_wire_rc` to get the RC when the corner is specified). We can see that there is a small difference in the values.
> 
> With this difference the `set_wire_rc` command will see differnt RC values on the 2 runs.

Thanks for debugging. 

> Also I was not able to reproduce this warnings.

Yes this isn't part of the test case. But I noticed that I didn't describe how to fully create it. If you call `set_wire_rc` with the argument `-corner` and *not* use `set_layer_rc `, you get the error above. 

Here is a tcl script to use [script.tar.gz](https://github.com/user-attachments/files/18344525/script.tar.gz)

The reason why I was reading the default values and setting them again is due to this error

arthurjolo, 2025-01-16:
I see, this error happens because if you call `set_wire_rc` and use `-corner` the command will try to get the rc values for that corner, and to set the rc values for a specific corner I believe you must use `set_layer_rc` command. Maybe some support could be added so that when `set_layer_rc` was not called to use the default values, but that is the same behavior as not using `-corner` and then it will use the default value and set the same wire_rc for all corners. @maliberty do you have any opinion on this, should we add this support?

maliberty, 2025-01-16:
I don't understand the linkage between set_layer_rc and set_wire_rc.  They seem like independent commands and information.  Would you expand on the connection?

arthurjolo, 2025-01-16:
Sure, when `set_wire_rc` is used with `-layer or -layers` the command will set the wires rc to be the same as the layer or the layers that are passed. If there is also `-corner` it will set the wire rc for that specific corner as the rc for the passed layer or the layers on that specific corner, if the layer rc for the corner hasn't been set it will set the wire rc as 0. To set the layer rc for a specific corner I think the only option is `set_layer_rc`.

kareefardi, 2025-01-19:
Thanks @arthurjolo for looking into it. I am interested in the current behavior of OR. Assuming the following scenarios:

Scenario A:
- Multiple corners are defined
- `set_layer_rc` is used for each corner
- `set_wire_rc` is used but _*without*_ passing `-corner`

Scenario B:
- Multiple corners are defined
- `set_wire_rc` is used but _*without*_ passing `-corner`

I would assume that calling `set_wire_rc` without passing `-corner` enforces the configuration across all defined corners. Would that be correct?

arthurjolo, 2025-03-04:
Hi @kareefardi, sorry for the late response your message got lost in my emails.

maliberty, 2025-06-01:
Is this resolved?

closed 2026-06-14 by github-actions[bot]