The-OpenROAD-Project/OpenROAD #6177: Make OpenROAD icon easier to see

oharboe, 2024-11-17:
### Description

The OpenROAD icon distinguishes it, but it is pretty much a mushy green blob on a grey background on Ubuntu:

[image]

Some other icons for comparison:

[image]

### Suggested Solution

Fix color scheme.

maliberty, 2024-11-17:
What size is the icon on your screen in pixels?

oharboe, 2024-11-17:
> What size is the icon on your screen in pixels?

See image below for details(screenshot), but ca. 50x40.

[image]

maliberty, 2024-11-18:
The icon has a transparent background so the window manager must be choosing the gray color.  How does it look if you use
[image]

oharboe, 2024-11-18:
> The icon has a transparent background so the window manager must be choosing the gray color.

That's a .jpg file, how do I "try" it?

I'll try latest master and see if I can tweak icon.png a bit.

oharboe, 2024-11-18:
An idea I was going to try, but [I still get no icon](https://github.com/The-OpenROAD-Project/OpenROAD/issues/6173#issuecomment-2482021927) when I try with master

[image]

maliberty, 2024-11-18:
> That's a .jpg file, how do I "try" it?

Replace the src/gui/resources/icon.png and make install should do it

oharboe, 2024-11-18:
> > That's a .jpg file, how do I "try" it?
> 
> Replace the src/gui/resources/icon.png and make install should do it

But it is a .jpg? icon.png is a png... I thought .jpeg did not encode transparency.

I reopend https://github.com/The-OpenROAD-Project/OpenROAD/issues/6173#issuecomment-2482021927 as I don't get an icon.

oharboe, 2024-11-18:
Example from https://github.com/The-OpenROAD-Project

[image]

closed 2024-11-18 by maliberty