The-OpenROAD-Project/OpenROAD #6186: Data races in `ant` module

eszpotanski, 2024-11-18:
### Describe the bug

When running `5_1_grt` stage for Ibex, ThreadSanitizer detects a few data races, usually in `ant::AntennaChecker`.

### Expected Behavior

Successful run without data races

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.11.5-arch1-1
os: Arch Linux
cmake version 3.30.5
-- The CXX compiler identification is GNU 14.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17013-gf7f634f88
-- System name: Linux
-- Compiler: GNU 14.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 187 lines cut ...]
    #3 std::_Rb_tree<odb::dbNet*, std::pair<odb::dbNet* const, ant::ViolationReport>, std::_Select1st<std::pair<odb::dbNet* const, ant::ViolationReport> >, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::_M_put_node(std::_Rb_tree_node<std::pair<odb::dbNet* const, ant::ViolationReport> >*) /usr/include/c++/12/bits/stl_tree.h:565 (openroad+0x3738724)
    #4 std::_Rb_tree<odb::dbNet*, std::pair<odb::dbNet* const, ant::ViolationReport>, std::_Select1st<std::pair<odb::dbNet* const, ant::ViolationReport> >, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::_M_drop_node(std::_Rb_tree_node<std::pair<odb::dbNet* const, ant::ViolationReport> >*) /usr/include/c++/12/bits/stl_tree.h:632 (openroad+0x3738724)
    [...]
    #37 tclAppInit /OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:507 (openroad+0xd0ce43)
    #38 ord::tclInit(Tcl_Interp*) /OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:550 (openroad+0xd0d9ae)
    #39 ord::tclAppInit(Tcl_Interp*) /OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:545 (openroad+0xd0d9ae)
    #40 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0)
    #41 __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58 (libc.so.6+0x27249)

  Previous read of size 8 at 0x7b1400be1230 by thread T15:
    #0 std::_Rb_tree<odb::dbNet*, std::pair<odb::dbNet* const, ant::ViolationReport>, std::_Select1st<std::pair<odb::dbNet* const, ant::ViolationReport> >, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::_S_left(std::_Rb_tree_node_base*) /usr/include/c++/12/bits/stl_tree.h:780 (openroad+0x374adfb)
    #1 std::_Rb_tree<odb::dbNet*, std::pair<odb::dbNet* const, ant::ViolationReport>, std::_Select1st<std::pair<odb::dbNet* const, ant::ViolationReport> >, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::_M_lower_bound(std::_Rb_tree_node<std::pair<odb::dbNet* const, ant::ViolationReport> >*, std::_Rb_tree_node_base*, odb::dbNet* const&) /usr/include/c++/12/bits/stl_tree.h:1952 (openroad+0x374adfb)
    #2 std::_Rb_tree<odb::dbNet*, std::pair<odb::dbNet* const, ant::ViolationReport>, std::_Select1st<std::pair<odb::dbNet* const, ant::ViolationReport> >, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::lower_bound(odb::dbNet* const&) /usr/include/c++/12/bits/stl_tree.h:1270 (openroad+0x374adfb)
    #3 std::map<odb::dbNet*, ant::ViolationReport, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::lower_bound(odb::dbNet* const&) /usr/include/c++/12/bits/stl_map.h:1307 (openroad+0x374adfb)
    #4 std::map<odb::dbNet*, ant::ViolationReport, std::less<odb::dbNet*>, std::allocator<std::pair<odb::dbNet* const, ant::ViolationReport> > >::at(odb::dbNet* const&) /usr/include/c++/12/bits/stl_map.h:549 (openroad+0x374adfb)
    #5 ant::AntennaChecker::checkGates(odb::dbNet*, bool, bool, std::basic_ofstream<char, std::char_traits<char> >&, odb::dbMTerm*, float, std::map<odb::dbITerm*, std::map<odb::dbTechLayer*, ant::NodeInfo, std::less<odb::dbTechLayer*>, std::allocator<std::pair<odb::dbTechLayer* const, ant::NodeInfo> > >, std::less<odb::dbITerm*>, std::allocator<std::pair<odb::dbITerm* const, std::map<odb::dbTechLayer*, ant::NodeInfo, std::less<odb::dbTechLayer*>, std::allocator<std::pair<odb::dbTechLayer* const, ant::NodeInfo> > > > > >&, std::vector<ant::Violation, std::allocator<ant::Violation> >&) /OpenROAD-flow-scripts/tools/OpenROAD/src/ant/src/AntennaChecker.cc:780 (openroad+0x3744c8f)
    #6 ant::AntennaChecker::checkNet(odb::dbNet*, bool, bool, std::basic_ofstream<char, std::char_traits<char> >&, odb::dbMTerm*, float, int&, int&, std::vector<ant::Violation, std::allocator<ant::Violation> >&) /OpenROAD-flow-scripts/tools/OpenROAD/src/ant/src/AntennaChecker.cc:1014 (openroad+0x3747f4c)
    #7 ant::AntennaChecker::checkAntennas(odb::dbNet*, int, bool) [clone ._omp_fn.0] /OpenROAD-flow-scripts/tools/OpenROAD/src/ant/src/AntennaChecker.cc:1122 (openroad+0x37487ab)
    #8 <null> <null> (libgomp.so.1+0x1cd8d)
[...]
```

### Additional Context

As ThreadSanitizer warnings can be quite long, logs are attached as file: [5_1_grt.log](https://github.com/user-attachments/files/17803129/ibex_tsan_5_1_grt.log)

luis201420, 2024-12-18:
Most of the issue has been resolved with the changes in #6364 . However, there is still a remaining problem that needs to be addressed. I  currently have two similar errors reported by **ThreadSanitizer**, like the one below, that I’m still unsure how to resolve.

````
  Read of size 4 at 0x7ffd32f78f8c by thread T19:
    #0 ant::AntennaChecker::checkAntennas(odb::dbNet*, int, bool) [clone ._omp_fn.0] /home/luis/OpenROAD-flow-scripts/tools/OpenROAD/src/ant/src/AntennaChecker.cc:1143 (openroad+0x3e0b6c3)
    #1 <null> <null> (libgomp.so.1+0x1dc0d)

  Previous write of size 4 at 0x7ffd32f78f8c by thread T21:
    #0 ant::AntennaChecker::checkAntennas(odb::dbNet*, int, bool) [clone ._omp_fn.0] /home/luis/OpenROAD-flow-scripts/tools/OpenROAD/src/ant/src/AntennaChecker.cc:1143 (openroad+0x3e0b6e1)
    #1 <null> <null> (libgomp.so.1+0x1dc0d)

  Location is stack of main thread.

  Location is global '<null>' at 0x000000000000 ([stack]+0x00000001ef8c)
````

I believe these errors are related to the variables in the following line (`net_violation_count` and `pin_violation_count`).

`#pragma omp parallel for schedule(dynamic) 
    reduction(+ : net_violation_count, pin_violation_count)`

I’ll keep investigating to confirm and resolve the issue.

closed 2024-12-20 by eder-matheus

luis201420, 2024-12-20:
The latest modifications made in the PR #6364  fix all the issues reported by ThreadSanitizer.

PR #6364 (2024-12-20): ant: fixing data races
This PR fixes data races found by ThreadSanitizer in ANT.  It is caused by the `net_to_report_` shared variable. 
The modifications made are:
- Adding mutex to protect access to the map.
- Removing unused parameters in functions.

Fixes #6186