The-OpenROAD-Project/OpenROAD #6197: MegaBoom global route failure wittled down from 2 hours to 10 minutes

oharboe, 2024-11-19:
untar https://drive.google.com/file/d/1LQ8fz3nJNmCcPdnlijQzfBXnqobIbkp2/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

Result after ca. 10 minutes(before deltaDebug.py this took 2 hours):

```

```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/megaboom/issues/205#issuecomment-2482216537_

eder-matheus, 2024-11-19:
@oharboe Looking at the 2 min test case, I think the issue is related to the routing layer adjustment used.

The regions near the macros would already have less resources due to the routing obstructions of the macro. When using 0.45 of adjustments, we are reducing these low resources even more, making these regions hard for the router.

0.45 seems to high for this design, so I would suggest try reducing it to 0.2, which was the value that I was able to finish the complete design.

Unfortunatelly, this configuration is sensible and tuning is necessary, but we have plans to make GRT smarter and not rely only in this configuration to handle congestion (see https://github.com/The-OpenROAD-Project/OpenROAD/issues/5423).

oharboe, 2024-11-19:
@eder-matheus Sounds good. I hope the test case is/was useful.

If this testcase is superfluous, then  close this issue?

closed 2024-11-19 by eder-matheus