The-OpenROAD-Project/OpenROAD #6206: M1 MacOS sequoia, openroad coredumps on exit

stefanottili, 2024-11-20:
### Describe the bug

openroad coredumps on exit

### Expected Behavior

no coredump

### Environment

```shell
M1 MacOS Sequoia, homebrew
```

### To Reproduce

openroad
exit

### Relevant log output

```shell
OpenROAD v2.0-17137-g6a3bf02d9 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
libc++abi: terminating due to uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 received
Stack trace:
 0# handler(int) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# pthread_kill in /usr/lib/system/libsystem_pthread.dylib
 3# abort in /usr/lib/system/libsystem_c.dylib
 4# __cxxabiv1::__aligned_malloc_with_fallback(unsigned long) in /usr/lib/libc++abi.dylib
 5# demangling_terminate_handler() in /usr/lib/libc++abi.dylib
 6# _objc_terminate() in /usr/lib/libobjc.A.dylib
 7# std::__terminate(void (*)()) in /usr/lib/libc++abi.dylib
 8# std::terminate() in /usr/lib/libc++abi.dylib
 9# fmt::v8::basic_memory_buffer<char, 500ul, std::__1::allocator<char>>::grow(unsigned long) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# ord::OpenRoad::~OpenRoad() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# ord::Tech::~Tech() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::~unique_ptr[abi:ne180100]() in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# __cxa_finalize_ranges in /usr/lib/system/libsystem_c.dylib
14# exit in /usr/lib/system/libsystem_c.dylib
15# gui::startGui(int&, char**, Tcl_Interp*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool, bool, bool) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# ord::tclInit(Tcl_Interp*) in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# Tcl_MainEx in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
18# Tcl_GetCurrentNamespace in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
19# main in OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
zsh: abort      openroad -gui
```

maliberty, 2024-11-20:
Are you using -gui?  I see gui::startGui in the stack.

stefanottili, 2024-11-20:
coredumps with/without -gui

maliberty, 2024-11-20:
I don't see how gui::startGui could get in the stack without -gui unless you have some odd stack corruption.  Can you build it debug and in the gdb use `catch throw` to see where the exception is generated?

stefanottili, 2024-11-20:
not a debug build, but should point to where the issue lies, no ?
```
OpenROAD v2.0-17139-gf827c1045 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
% exit
libc++abi: terminating due to uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Process 13538 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x0000000198fc2600 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`__pthread_kill:
->  0x198fc2600 <+8>:  b.lo   0x198fc2620    ; <+40>
    0x198fc2604 <+12>: pacibsp 
    0x198fc2608 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x198fc260c <+20>: mov    x29, sp
Target 0: (openroad) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x0000000198fc2600 libsystem_kernel.dylib`__pthread_kill + 8
    frame #1: 0x0000000198ffaf70 libsystem_pthread.dylib`pthread_kill + 288
    frame #2: 0x0000000198f07908 libsystem_c.dylib`abort + 128
    frame #3: 0x0000000198fb144c libc++abi.dylib`abort_message + 132
    frame #4: 0x0000000198f9fa24 libc++abi.dylib`demangling_terminate_handler() + 320
    frame #5: 0x0000000198c453f4 libobjc.A.dylib`_objc_terminate() + 172
    frame #6: 0x0000000198fb0710 libc++abi.dylib`std::__terminate(void (*)()) + 16
    frame #7: 0x0000000198fb06b4 libc++abi.dylib`std::terminate() + 108
    frame #8: 0x000000010002156c openroad`__clang_call_terminate + 16
    frame #9: 0x0000000100030a3c openroad`ord::OpenRoad::~OpenRoad() + 248
    frame #10: 0x0000000100030628 openroad`ord::Tech::~Tech() + 28
    frame #11: 0x0000000100034434 openroad`std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::~unique_ptr[abi:ne180100]() + 32
    frame #12: 0x0000000198eb6998 libsystem_c.dylib`__cxa_finalize_ranges + 476
    frame #13: 0x0000000198eb675c libsystem_c.dylib`exit + 44
    frame #14: 0x0000000104c85468 libtcl8.6.dylib`Tcl_Exit + 136
    frame #15: 0x0000000104c1bd9c libtcl8.6.dylib`Tcl_ExitObjCmd + 116
    frame #16: 0x0000000104c106f4 libtcl8.6.dylib`TclNRRunCallbacks + 80
    frame #17: 0x0000000104c97a90 libtcl8.6.dylib`Tcl_RecordAndEvalObj + 476
    frame #18: 0x0000000104cb72ac libtcl8.6.dylib`Tcl_MainEx + 1516
    frame #19: 0x0000000104cb79a0 libtcl8.6.dylib`Tcl_Main + 52
    frame #20: 0x0000000100034830 openroad`main + 952
    frame #21: 0x0000000198c78274 dyld`start + 2840
(lldb) 
```

maliberty, 2024-11-20:
Can you break when the exception is thrown?  "terminating due to uncaught exception" suggests it was already thrown.

maliberty, 2024-11-20:
In gdb `catch throw` will make it stop on throw.

stefanottili, 2024-11-20:
https://stackoverflow.com/questions/8122375/lldb-breakpoint-on-exceptions-equivalent-of-gdbs-catch-throw
claims to have the lldb equivalent to `catch throw` but it doesn't seem to add any information
```
lldb ../../install/OpenROAD/bin/openroad
(lldb) break set -E C++
Breakpoint 2: 2 locations.
(lldb) run
There is a running process, kill it and restart?: [Y/n] y
Process 39126 exited with status = 9 (0x00000009) killed
Process 39140 launched: 'OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad' (arm64)
OpenROAD v2.0-17139-gf827c1045 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
% exit
Process 39140 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1 2.1
    frame #0: 0x0000000198fb3c28 libc++abi.dylib`__cxa_throw
libc++abi.dylib`__cxa_throw:
->  0x198fb3c28 <+0>:  pacibsp 
    0x198fb3c2c <+4>:  stp    x22, x21, [sp, #-0x30]!
    0x198fb3c30 <+8>:  stp    x20, x19, [sp, #0x10]
    0x198fb3c34 <+12>: stp    x29, x30, [sp, #0x20]
Target 0: (openroad) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1 2.1
  * frame #0: 0x0000000198fb3c28 libc++abi.dylib`__cxa_throw
    frame #1: 0x0000000198f30a44 libc++.1.dylib`std::__1::__throw_system_error[abi:ne180100](std::__1::error_code, char const*) + 92
    frame #2: 0x0000000198f309e8 libc++.1.dylib`std::__1::__throw_system_error(int, char const*) + 48
    frame #3: 0x0000000198f32770 libc++.1.dylib`std::__1::mutex::lock() + 40
    frame #4: 0x0000000101be80cc openroad`odb::dbDatabase::destroy(odb::dbDatabase*) + 28
    frame #5: 0x00000001000309cc openroad`ord::OpenRoad::~OpenRoad() + 152
    frame #6: 0x0000000100030618 openroad`ord::Tech::~Tech() + 28
    frame #7: 0x0000000100034424 openroad`std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::~unique_ptr[abi:ne180100]() + 32
    frame #8: 0x0000000198eb6998 libsystem_c.dylib`__cxa_finalize_ranges + 476
    frame #9: 0x0000000198eb675c libsystem_c.dylib`exit + 44
    frame #10: 0x0000000104c85468 libtcl8.6.dylib`Tcl_Exit + 136
    frame #11: 0x0000000104c1bd9c libtcl8.6.dylib`Tcl_ExitObjCmd + 116
    frame #12: 0x0000000104c106f4 libtcl8.6.dylib`TclNRRunCallbacks + 80
    frame #13: 0x0000000104c97a90 libtcl8.6.dylib`Tcl_RecordAndEvalObj + 476
    frame #14: 0x0000000104cb72ac libtcl8.6.dylib`Tcl_MainEx + 1516
    frame #15: 0x0000000104cb79a0 libtcl8.6.dylib`Tcl_Main + 52
    frame #16: 0x0000000100034820 openroad`main + 952
    frame #17: 0x0000000198c78274 dyld`start + 2840
(lldb) 
```

maliberty, 2024-11-20:
At least gui is out of the stack.  It still makes no sense to me though as Tech creates an OpenRoad which creates a dbDatabase.  ~Tech call ~OpenRoad which destroys the database.  I don't see an obvious bug and I have no way to reproduce and debug it myself.

maliberty, 2024-11-20:
My only guess is `  std::lock_guard<std::mutex> lock(db_tbl_mutex);` in dbDatabase::destroy is somehow unhappy.  If you comment it out does the crash disappear?

maliberty, 2024-11-21:
Perhaps it is an order of destruction thing,
```
static std::unique_ptr<ord::Tech> the_tech;
static std::mutex db_tbl_mutex;
```
are both cleaned up at program exit.  If the mutex is destroyed first then the database destroy can't use the mutex.

maliberty, 2024-11-21:
Another experiment is to add to clean these up after Tcl_Main (add the middle two lines)
```
  Tcl_Main(1, argv, ord::tclAppInit);
  the_design.reset();
  the_tech.reset();
  return 0;

```

stefanottili, 2024-11-21:
```
   Tcl_Main(1, argv, ord::tclAppInit);
+  the_design.reset();
+  the_tech.reset();
   return 0;
```
doesn't seem to help, here the stack trace from the debug build.
```
src/openroad
OpenROAD v2.0-17139-gf827c1045 
Features included (+) or not (-): +Charts +GPU +GUI +Python : DEBUG
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
% exit
libc++abi: terminating due to uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 received
Stack trace:
 0# handler(int) in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# pthread_kill in /usr/lib/system/libsystem_pthread.dylib
 3# abort in /usr/lib/system/libsystem_c.dylib
 4# __cxxabiv1::__aligned_malloc_with_fallback(unsigned long) in /usr/lib/libc++abi.dylib
 5# demangling_terminate_handler() in /usr/lib/libc++abi.dylib
 6# _objc_terminate() in /usr/lib/libobjc.A.dylib
 7# std::__terminate(void (*)()) in /usr/lib/libc++abi.dylib
 8# std::terminate() in /usr/lib/libc++abi.dylib
 9# std::__1::__atomic_base<int, false>::load[abi:ne180100](std::__1::memory_order) const in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
10# ord::OpenRoad::~OpenRoad() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
11# ord::OpenRoad::~OpenRoad() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
12# ord::Tech::~Tech() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
13# ord::Tech::~Tech() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
14# std::__1::default_delete<ord::Tech>::operator()[abi:ne180100](ord::Tech*) const in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
15# std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::reset[abi:ne180100](ord::Tech*) in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
16# std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::~unique_ptr[abi:ne180100]() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
17# std::__1::unique_ptr<ord::Tech, std::__1::default_delete<ord::Tech>>::~unique_ptr[abi:ne180100]() in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
18# __cxa_finalize_ranges in /usr/lib/system/libsystem_c.dylib
19# exit in /usr/lib/system/libsystem_c.dylib
20# Tcl_Finalize in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
21# Tcl_ExitObjCmd in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
22# TclNRRunCallbacks in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
23# Tcl_RecordAndEvalObj in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
24# Tcl_MainEx in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
25# Tcl_GetCurrentNamespace in /opt/homebrew/Cellar/tcl-tk@8/8.6.15/lib/libtcl8.6.dylib
26# main in OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
zsh: abort      src/openroad
```
If there are no other ideas, I'll have to resort to `git bisect`, this used to work just days ago.

stefanottili, 2024-11-21:
git bisect points to 4099a0e5840721f4d56b54d431d1195cd30dcd9d as the first bad commit
`odb: put a lock around db_tcl and use atomic for db_unique_id` 
The previous f2765f209925d5827e89a7d128337ff159371a8c works.

maliberty, 2024-11-21:
I'm not surprised that is the commit, I just don't see the error.  With the suggested change please give the stack when the exception is thrown.

stefanottili, 2024-11-21:
I already did ...

stefanottili, 2024-11-21:
```
691 void dbDatabase::destroy(dbDatabase* db_)
692 {
693 //  std::lock_guard<std::mutex> lock(db_tbl_mutex);
694   _dbDatabase* db = (_dbDatabase*) db_;
695   db_tbl->destroy(db);
696 }
```
commenting out line 693 get's rid of the coredump

maliberty, 2024-11-21:
The stack trace from the debug build shows "terminating due to uncaught exception" which means you didn't stop when it was thrown.

stefanottili, 2024-11-21:
yes, I have not figured out how to do that with lldb ....

maliberty, 2024-11-21:
"Tcl_Main does not return." Ugh - wait a brain dead language.  It never reaches that code.  I'll try out some ideas.

maliberty, 2024-11-21:
@stefanottili please try #6209

stefanottili, 2024-11-21:
@maliberty works !

commit d3e7fec9 (2024-11-21): odb: ensure db_tbl_mutex is not destroyed.

Fixes #6206
closed 2024-11-21 by maliberty