The-OpenROAD-Project/OpenROAD #6210: Incorrect error message: Increase the region 0.00um-7.01um on the LEFT edge from 7.01um to at least 6.82um.

oharboe, 2024-11-21:
### Describe the bug

untar [io-placement-error.tar.gz](https://github.com/user-attachments/files/17845317/io-placement-error.tar.gz)

```
$ ./run-me-sdq_17x64-asap7-base.sh 
OpenROAD v2.0-17048-g82450fd6f 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PPL-0048] Restrict pins [ R0_addr[0] R0_addr[1] R0_addr[2] R0_addr[3] R0_addr[4] ... ] to region 0.00u-7.01u at the LEFT edge.
place_pins -hor_layers M4 -ver_layers M5 -min_distance 1 -min_distance_in_tracks -random
Found 0 macro blocks.
[WARNING PPL-0110] Constraint has 142 pins, but only 131 available slots.
Increase the region 0.00um-7.01um on the LEFT edge from 7.01um to at least 6.82um.
[ERROR PPL-0111] 1 constraint(s) does not have available slots for the pins.
Error: io_placement_util.tcl, 11 PPL-0111
openroad> 

```

### Expected Behavior

The correct minimum value to increase the region to

### Environment

```shell
OpenROAD v2.0-17048-g82450fd6f
```

### To Reproduce

See above

eder-matheus, 2024-11-21:
@oharboe This is a tricky case. PPL has a default offset from the corners of the die of 1 micron. When you defined your constraint region from 0 to 7.01, you actually have your region from 1 to 7.01. This default offset is used to avoid pins near the corner that could make it harder to route.

Perhaps adding an info message about this offset can help, but I'm not sure what is the best path here regarding the incorrect error message. Perhaps showing the actual useful size of it?

oharboe, 2024-11-21:
So 0-7.01 is automatically converted to 1-7.01? 

I think that the message could just give the range, increase from 1-7.01 to 1-7.82 + document this behavior...

Also, I tried to write some .tcl that calculated the range I need to provide fit the pins into the lower left edge or the left of the bottom edge.

Any tips on how to write this code?

```
source $::env(SCRIPTS_DIR)/util.tcl
source util.tcl

# By placing the pins at the left edge, then by mirroring on
# X and Y, it is possible to place the pins in 4 significantly
# different orientations. In lieu of being able to rotate a
# macro 90 degrees, this is a good compromise.
set db [::ord::get_db]
set block [[$db getChip] getBlock]
set die_bbox [$block getCoreArea]
set dbu_per_uu [expr double([[$db getTech] getDbUnitsPerMicron])]
set scale [expr 1 / $dbu_per_uu]
#puts $f "width: [expr [$die_bbox xMax] * $scale]"
#puts $f "height: [expr [$die_bbox yMax] * $scale]"
set height [expr [$die_bbox yMax] * $scale]
set width [expr [$die_bbox xMax] * $scale]

set tech [$block getTech]
if {$height > $width} {
  set layer [$tech findLayer $::env(IO_PLACER_V)]
  set pitch [expr [$layer getPitchX] * $scale]
  set edge_length $height
  set edge left
} else {
  set layer [$tech findLayer $::env(IO_PLACER_H)]
  set pitch [expr [$layer getPitchY] * $scale]
  set edge_length $width
  set edge top
}
set pin_names [match_pins {(R|W)[0-9]+_.*}]

log_cmd set_io_pin_constraint -region $edge:0-[expr min($edge_length, $pitch * ([llength $pin_names] + 4) * 1.3)] -pin_names $pin_names
```

oharboe, 2024-11-22:
@eder-matheus For me this is low priority as I ran into it during an experiment that didn't pan out https://github.com/The-OpenROAD-Project/megaboom/pull/214

closed 2024-11-26 by eder-matheus