The-OpenROAD-Project/OpenROAD #6225: Power simulation based on global route gives error message

oharboe, 2024-11-24:
With https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2587, the below should:

```
make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk grt
make ABSTRACT_SOURCE=5_1_grt DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk do-generate_abstract
cp results/asap7/mock-array_Element/base/5_1_grt.v results/asap7/mock-array_Element/base/6_final.v
cp results/asap7/mock-array_Element/base/5_1_grt_default.spef results/asap7/mock-array_Element/base/6_final.spef
make ABSTRACT_SOURCE=5_1_grt DESIGN_CONFIG=designs/asap7/mock-array/config.mk grt
cp results/asap7/mock-array/base/5_1_grt.v results/asap7/mock-array/base/6_final.v
cp results/asap7/mock-array/base/5_1_grt_default.spef results/asap7/mock-array/base/6_final.spef
cp results/asap7/mock-array/base/5_1_grt.sdc results/asap7/mock-array/base/6_final.sdc
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk simulate power
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6219#discussioncomment-11359461_

Error:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk power
```

.spef files generated by estimate_parasitics -global_route do not not match global route .v file:

```
[deleted]
Warning: ./results/asap7/mock-array/base/6_final.spef line 4182, pin REG_9$_DFF_P_/QN not found.
Warning: ./results/asap7/mock-array/base/6_final.spef line 4183, pin _065_/A not found.
Warning: ./results/asap7/mock-array/base/6_final.spef line 4196, pin _065_/A not found.
Warning: ./results/asap7/mock-array/base/6_final.spef line 4197, pin REG_9$_DFF_P_/QN not found.
Warning: ./results/asap7/mock-array/base/6_final.spef line 4201, pin REG_0$_DFF_P_/QN not found.
[deleted]

```

oharboe, 2024-11-24:
Examining the .spef file from `make final` vs `make grt`, there are some differences I see immediately:

- `estimate_parasitics -spec_file` does not use a `NAME_MAP`
- The syntax to refer to the pin is different `:QN`, works with OpenSTA, vs `/QN`, which does not appear to work: `*I REG_9\$_DFF_P_/QN O *D DFFHQNx2_ASAP7_75t_R` vs `*I *23402:QN O *D DFFHQNx2_ASAP7_75t_R`

`estimate_parasitics -spec_file` version:

```spef
*D_NET _000_ 0.804904
*CONN
*I REG_9\$_DFF_P_/QN O *D DFFHQNx2_ASAP7_75t_R
*I _065_/A I *D INVx3_ASAP7_75t_R
*CAP
1 _000_:1 0.0589401
2 _000_:2 0.0362502
3 _000_:3 0.313907
4 _000_:4 0.313907
5 _000_:5 0.0296043
*RES
1 _000_:1 _000_:2 0.0130799
2 _000_:2 _000_:3 0.0172
3 _000_:3 _000_:4 0.117719
4 _000_:4 _000_:5 0.0172
5 _065_/A _000_:5 0.0278819
6 REG_9\$_DFF_P_/QN _000_:1 0.025387
*END
```

vs:

```
[deleted]
*NAME_MAP
*23402 REG_9\$_DFF_P_
[deleted]
*D_NET *1 0.000501571
*CONN
*I *23403:A I *D INVx3_ASAP7_75t_R
*I *23402:QN O *D DFFHQNx2_ASAP7_75t_R
*CAP
1 *23403:A 0
2 *23402:QN 3.26469e-05
3 *1:10 0.00019743
4 *1:8 0.000230077
5 *1:8 *15540:21 6.7842e-07
6 *1:8 *19703:6 0
7 *1:10 *8:8 0
8 *1:10 *15510:21 0
9 *1:10 *15510:27 0
10 *1:10 *18132:8 1.34466e-06
11 *1:10 *19662:8 5.37866e-07
12 *1:10 *19673:8 0
13 *1:10 *19712:12 1.41197e-05
14 *1:10 *22281:21 3.05118e-08
15 *1:10 *22286:27 1.24158e-05
16 *1:10 *22295:27 9.30043e-06
17 *1:10 *22297:27 1.93482e-06
18 *1:10 *22327:27 0
19 *1:10 *22336:27 3.05118e-08
20 *1:10 *22341:27 1.02359e-06
*RES
1 *23402:QN *1:8 47.7706 
2 *1:8 *1:10 144.171 
3 *1:10 *23403:A 34.4 
*END
```

oharboe, 2024-12-05:
@maliberty Will I be able to save spef files from cts using `estimate_parasitics -placement`?

Skipping global route and detailed route could be handy in early exploration of power or at least when running through the CI flow.

oharboe, 2024-12-11:
@eder-matheus Purely for my planning purposes, do you have an ETA on this?

We're having problems with global routing/macro placement(reported to @maliberty as I can't share the design).

I we use .spef files from cts(estimate_parasitics -placement -spef_file), I would be able to do OpenSTA power simulations while I'm waiting for global routing/macro placement issues to be sorted out.

eder-matheus, 2024-12-11:
> [@eder-matheus](https://github.com/eder-matheus) Purely for my planning purposes, do you have an ETA on this?

Yes, I'm planning to finish it at most by tomorrow.

oharboe, 2024-12-12:
We have a big testcase ready....

eder-matheus, 2024-12-12:
@oharboe Could you try this branch: https://github.com/The-OpenROAD-Project/OpenROAD/pull/6346?

This is a quick fix that should work for your case, but I still working in a more robust solution to have it saved in odb.

oharboe, 2024-12-12:
> @oharboe Could you try this branch: #6346?
> 
> This is a quick fix that should work for your case, but I still working in a more robust solution to have it saved in odb.

@nekronos ?

eder-matheus, 2024-12-16:
@oharboe I've merged my quick fix to the master branch. Let me know if you still have issues.

oharboe, 2024-12-16:
close?

this will be fixed in a more maintainable way, but the ask in this issue is addressed.

I will open a new issue if I have new problems.

oharboe, 2024-12-19:
@eder-matheus It is still failing in a non-public testcase I have shared with @maliberty 

Excerpt from the testcase to give you a sense of what is going on:

In this test-case `estimate_parasitics -placement` is used from cts.

```
ODB_FILE=.../base/4_cts.odb make run RUN_SCRIPT=parasitics.tcl
```

parasitics.tcl:

```tcl
source $::env(SCRIPTS_DIR)/open.tcl
estimate_parasitics -placement -spef_file $::env(RESULTS_DIR)/4_cts.spef
exec mv $::env(RESULTS_DIR)/4_cts_default.spef $::env(RESULTS_DIR)/4_cts.spef
write_verilog $::env(RESULTS_DIR)/4_cts.v
```

Here are some snippets from that test-case:

[image]
[image]

```
OpenSTA 2.3.2 895ab26f6c Copyright (c) 2021, Parallax Software, Inc.
[deleted]
Warning: .../base/4_cts.spef line 4179, instance branchPredictor/_3206_:A not found.
```

eder-matheus, 2024-12-19:
@oharboe Just to clarify, what is the instance name in the second figure? Is it just `branchPredictor` and the pin name is `_3206_/A`?

oharboe, 2024-12-19:
> @oharboe Just to clarify, what is the instance name in the second figure? Is it just `branchPredictor` and the pin name is `_3206_/A`?

See Verilog snippet above. The instance name, the name of the INV is visible there.

oharboe, 2024-12-19:
@eder-matheus How far away is writing parasitics to .odb and using a single write spef implementation?

eder-matheus, 2024-12-19:
> @eder-matheus How far away is writing parasitics to .odb and using a single write spef implementation?

I'll prioritize it to have it done for the next week. I had some other things in my list, but since the quick fix looks insufficient I'll give spend more time on the odb approach.

maliberty, 2024-12-23:
I've put in a fix for the current method that allows the private test case to load.

oharboe, 2024-12-23:
The fix is tested and merged.

Close?