The-OpenROAD-Project/OpenROAD #6227: Segmentation Fault in OpenROAD During Multi-Corner STA (sky130A PDK)

weafo, 2024-11-24:
### Describe the bug

OpenROAD crashes with a segmentation fault during the execution of the multi-corner static timing analysis (STA) step in OpenLane. The crash occurs when using the sky130A PDK, and the log reports a segmentation violation in libc. The issue is reproducible across multiple runs with the same configuration.

Steps I do:
1. Clone the OpenLane repository and set up the environment.
2. Install the sky130A PDK and set the `PDK_ROOT` environment variable.
3. Run the following command:
   ./flow.tcl -design <your_design> -run_full_flow
4. The segmentation fault occurs during the STA step in OpenROAD.

### Expected Behavior

OpenROAD should complete the multi-corner STA step without crashing and generate timing reports.

### OpenROAD Environment

```shell
I am using OpenROAD through the OpenLane Docker image: efabless/openlane-tools:openroad_app-cdf98f3571ef892d091b6825c5d473bcfbf6e167-centos-7-amd64 with OpenROAD.
```

### OpenLane Environment

```shell
open_pdks cdf98f3571ef892d091b6825c5d473bcfbf6e167
Kernel: 5.15.0-122-generic
Description: Ubuntu 20.04.6 LTS
Docker version 27.3.1, build ce12230
Python 3.11.10
cdf98f35 Thu Jul 18 14:28:46 2024 +0300 Propagate the clock in some STA reports (#2134)
python-venv: INSTALLED
```

### To Reproduce

1. Clone the OpenLane repository.
2. Install the sky130A PDK and set PDK_ROOT to point to the installation.
3. Navigate to the OpenLane directory.
4. run `make mount`
5. Run the following command:
   ```
   ./flow.tcl -design adder_depth_2023 -init_design_config -add_to_designs
   ./flow.tcl -design adder_depth_2023 -init_design_config -add_to_designs -config_file config.tcl
   ./flow.tcl -design adder_depth_2023 ```
6. The segmentation fault occurs during the multi-corner STA step in OpenROAD.

I've also attached a zip of the issue reproducible folder.
[issue_reproducible.zip](https://github.com/user-attachments/files/17893909/issue_reproducible.zip)

### Relevant log output

```shell
OpenLane cdf98f3571ef892d091b6825c5d473bcfbf6e167
All rights reserved. (c) 2020-2023 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/adder_depth_2023/config.tcl'...
[INFO]: Process Design Kit: sky130A
[INFO]: PDK Root: /home/mvguser/.volare
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/adder_depth_2023/runs/RUN_2024.11.24_14.53.20
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[INFO]: Running linter (Verilator) (log: designs/adder_depth_2023/runs/RUN_2024.11.24_14.53.20/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[... 100 lines cut ...]
         ^^^^^^^^^^^^^^^^
  File "/nix/store/vr8dx2vs7nhz4cz2vfyypx69vbwwrmk5-python3-3.11.9-env/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/vr8dx2vs7nhz4cz2vfyypx69vbwwrmk5-python3-3.11.9-env/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/openlane/scripts/generate_reports.py", line 51, in cli
    report = Report(design, tag, design_name, params, run_path).get_report()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/openlane/scripts/report/report.py", line 809, in get_report
    *self.extract_all_values(),
     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/openlane/scripts/report/report.py", line 359, in extract_all_values
    power_report = Artifact(
                   ^^^^^^^^^
  File "/openlane/scripts/report/report.py", line 62, in __init__
    self.pathname = matches[0]
                    ~~~~~~~^^^
IndexError: list index out of range
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Skipping CTS...
```

maliberty, 2024-11-24:
I can't find that image
```
Unable to find image 'efabless/openlane-tools:openroad_app-cdf98f3571ef892d091b6825c5d473bcfbf6e167-centos-7-amd64' locally
docker: Error response from daemon: manifest for efabless/openlane-tools:openroad_app-cdf98f3571ef892d091b6825c5d473bcfbf6e167-centos-7-amd64 not found: manifest unknown: manifest unknown.
```
what version of OpenROAD is in that image?  If it is not very current then please try with the head of OR first.

maliberty, 2024-11-24:
You need to make a self contained test case.  The steps to reproduce should be untar & run, not install OL.

weafo, 2024-11-24:
Sorry for the confusion, I am not using a separate Docker image of OpenROAD. Instead, I use it as a dependency within OpenLane. The version of OpenROAD in my setup is: OpenROAD b16bda7e82721d10566ff7e2b68f1ff0be9f9e38.

maliberty, 2024-11-25:
That commit is from Jun 3.  Please try with the head of OR.

weafo, 2024-12-04:
I encountered an error while building OpenROAD with the head of OR.
here's the steps:
1. Edited `flake.nix` to update the `rev`
```
openroad = pkgs.openroad.override {
  rev = "e3ab3fb6e073ec23e499537350008fb2aaaead2e"; 
  sha256 = ""; # Temporarily empty
  openroad-abc = self.openroad-abc;
  opensta = self.opensta;
};
```
2. Ran the build command `nix build .#openroad --extra-experimental-features nix-command --extra-experimental-features flakes`
3.
```
error: hash mismatch in fixed-output derivation '/nix/store/qyp5zdcziw8arbp02g5ml8wpdba9gw0r-source.drv':
         specified: sha256-AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
            got:    sha256-cYt9Ua9gafWsukB+FBXCuA1leQoXJlDvvfdZtBwGRYI=
```
4. Updated flake.nix with the fetched sha256
```
openroad = pkgs.openroad.override {
  rev = "e3ab3fb6e073ec23e499537350008fb2aaaead2e"; 
  sha256 = "sha256-cYt9Ua9gafWsukB+FBXCuA1leQoXJlDvvfdZtBwGRYI="; 
  openroad-abc = self.openroad-abc;
  opensta = self.opensta;
};
```
5. Re-ran the build command `nix build .#openroad --extra-experimental-features nix-command --extra-experimental-features flakes`
```
warning: Git tree '/home/mvguser/hdd/OL/OpenLane' is dirty
warning: ignoring untrusted substituter 'https://openlane.cachix.org', you are not a trusted user.
Run `man nix.conf` for more information on the `substituters` configuration option.
warning: Ignoring setting 'auto-allocate-uids' because experimental feature 'auto-allocate-uids' is not enabled
warning: Ignoring setting 'impure-env' because experimental feature 'configurable-impure-env' is not enabled
error: builder for '/nix/store/k2d3pmaqgg8wmzmbkcqrd238cp4wbabh-openroad.drv' failed with exit code 2;
       last 25 log lines:
       > In file included from /build/source/src/gui/src/gui.cpp:46:
       > In file included from /build/source/src/gui/src/mainWindow.h:43:
       > /build/source/src/gui/src/findDialog.h:46:8: warning: 'accept' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
       >    46 |   void accept();
       >       |        ^
       > /nix/store/p8cr0k5g2jbjy6k80sdhfyr7vp4zzl8w-qtbase-5.15.12-dev/include/QtWidgets/qdialog.h:97:18: note: overridden virtual function is here
       >    97 |     virtual void accept();
       >       |                  ^
       > In file included from /build/source/src/gui/src/gui.cpp:46:
       > In file included from /build/source/src/gui/src/mainWindow.h:43:
       > /build/source/src/gui/src/findDialog.h:47:8: warning: 'reject' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
       >    47 |   void reject();
       >       |        ^
       > /nix/store/p8cr0k5g2jbjy6k80sdhfyr7vp4zzl8w-qtbase-5.15.12-dev/include/QtWidgets/qdialog.h:98:18: note: overridden virtual function is here
       >    98 |     virtual void reject();
       >       |                  ^
       > 7 warnings generated.
       > [ 47%] Building CXX object src/gui/CMakeFiles/gui.dir/src/globalConnectDialog.cpp.o
       > 1 error generated.
       > make[2]: *** [src/gui/CMakeFiles/gui.dir/build.make:436: src/gui/CMakeFiles/gui.dir/src/staGuiInterface.cpp.o] Error 1
       > make[2]: *** Waiting for unfinished jobs....
       > 2 warnings generated.
       > 2 warnings generated.
       > make[1]: *** [CMakeFiles/Makefile2:4448: src/gui/CMakeFiles/gui.dir/all] Error 2
       > make: *** [Makefile:136: all] Error 2
       For full logs, run 'nix-store -l /nix/store/k2d3pmaqgg8wmzmbkcqrd238cp4wbabh-openroad.drv'.
```
Since OpenROAD is used as a dependency, I'm not entirely sure how to properly update it without manually modifying flake.nix. Is there a better approach?

donn, 2024-12-04:
The steps @weafo took were correct, slightly undermined by how the latest version of OpenROAD doesn't build with the flake yet and I have not had the chance to submit a patch for it. Something about boost::optional and fmt.