The-OpenROAD-Project/OpenROAD #6235: Placer can't converge for new mpl2 IOs changes in sky130hd/uW

AcKoucher, 2024-11-25:
### Describe the bug

The last run in Secure-CI for the new IO Clusters in mpl2 (secure-mpl2-constraints) showed a failure in sky130hd/uW in which, during global placement skip io (3_1), the placer can't converge.

### Expected Behavior

The placer should converge or we should have a better understanding on what's causing a divergence if it is in fact a true one.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-1024-gcp
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17329-gc8f03bb7a
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 121 lines cut ...]
[NesterovSolve] Iter:  200 overflow: 0.650 HPWL: 5730633764
[NesterovSolve] Iter:  210 overflow: 0.613 HPWL: 5799011531
[NesterovSolve] Iter:  220 overflow: 0.575 HPWL: 5918421347
[NesterovSolve] Iter:  230 overflow: 0.528 HPWL: 6021712994
[NesterovSolve] Iter:  240 overflow: 0.487 HPWL: 6059579547
[NesterovSolve] Iter:  250 overflow: 0.442 HPWL: 6100101971
[NesterovSolve] Iter:  260 overflow: 0.397 HPWL: 6128859299
[NesterovSolve] Iter:  270 overflow: 0.352 HPWL: 6179198317
[NesterovSolve] Iter:  280 overflow: 0.315 HPWL: 6213107366
[NesterovSolve] Iter:  290 overflow: 0.283 HPWL: 6240217003
[NesterovSolve] Iter:  300 overflow: 0.254 HPWL: 6269842592
[NesterovSolve] Iter:  310 overflow: 0.226 HPWL: 6286669742
[NesterovSolve] Iter:  320 overflow: 0.206 HPWL: 6153622281
[NesterovSolve] Iter:  330 overflow: 0.185 HPWL: 6117815535
[NesterovSolve] Iter:  340 overflow: 0.165 HPWL: 6130453725
[NesterovSolve] Iter:  350 overflow: 0.148 HPWL: 6143974331
[NesterovSolve] Iter:  360 overflow: 0.134 HPWL: 6186392922
[NesterovSolve] Iter:  370 overflow: 0.152 HPWL: 6786347162
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: global_place_skip_io.tcl, 12 GPL-0307
Command exited with non-zero status 1
Elapsed time: 0:22.58[h:]min:sec. CPU time: user 324.61 sys 0.81 (1441%). Peak memory: 508292KB.
make[1]: *** [Makefile:676: do-3_1_place_gp_skip_io] Error 1
make: *** [Makefile:674: results/sky130hd/microwatt/base/3_1_place_gp_skip_io.odb] Error 2
```

gudeh, 2024-12-30:
From local tests with the latest changes in gpl, the branch mpl2-constraints did not present a divergence on stage 3-1 anymore. I am leaving the issue open because I could not merge mpl2-constraints branch with master due to conflicts:

```
git merge private/master 
Auto-merging src/mpl2/README.md
Auto-merging src/mpl2/include/mpl2/rtl_mp.h
Auto-merging src/mpl2/src/Mpl2Observer.h
Auto-merging src/mpl2/src/SimulatedAnnealingCore.cpp
Auto-merging src/mpl2/src/clusterEngine.cpp
Auto-merging src/mpl2/src/clusterEngine.h
Auto-merging src/mpl2/src/graphics.cpp
CONFLICT (content): Merge conflict in src/mpl2/src/graphics.cpp
Auto-merging src/mpl2/src/graphics.h
Auto-merging src/mpl2/src/hier_rtlmp.cpp
CONFLICT (content): Merge conflict in src/mpl2/src/hier_rtlmp.cpp
Auto-merging src/mpl2/src/hier_rtlmp.h
CONFLICT (content): Merge conflict in src/mpl2/src/hier_rtlmp.h
Auto-merging src/mpl2/src/mpl.i
Auto-merging src/mpl2/src/mpl.tcl
Auto-merging src/mpl2/src/object.h
Auto-merging src/mpl2/src/rtl_mp.cpp
CONFLICT (modify/delete): src/mpl2/test/regression_tests.tcl deleted in private/master and modified in HEAD.  Version HEAD of src/mpl2/test/regression_tests.tcl left in tree.
Automatic merge failed; fix conflicts and then commit the result.
```

AcKoucher, 2025-01-06:
I updated the branch in #5809 and ran uW locally. GPL converges so I'm closing this.

PS I now having problems with post drt:

```
[INFO DRT-0180] Post processing.
[INFO GRT-0012] Found 548 antenna violations.
[INFO GRT-0015] Inserted 1257 diodes.
[ERROR GRT-0232] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: detail_route.tcl, 61 GRT-0232

closed 2025-01-06 by AcKoucher