The-OpenROAD-Project/OpenROAD #6243: DRC dropdown shows .drc choices for other designs than the current one

oharboe, 2024-11-26:
### Description

This dropdown shows DRC files that do not belong to the loaded design. Very confusing. I lost some time to this behavior before I realized that it simply shows a list of recent DRC files regardless of which design it belongs to.

At this point, the dropdown is pretty much useless and more positively harmful: I can't trust what it shows me, almost exclusively is showing me something that is wrong for the currently loaded design.

This is especially devious when switching between versions and variants of the same design, there's no visible clue that the DRC file is simply belonging to a different version of the same design.

[image]

### Suggested Solution

Filter the list on DRC files that belong to the loaded design.

gadfort, 2024-11-26:
@oharboe the list shows all the markers stored in the database, so it's not clear to me how this list contains markers from a different design. Can you provide a test or steps to replicate.

oharboe, 2024-11-26:
> @oharboe the list shows all the markers stored in the database, so it's not clear to me how this list contains markers from a different design. Can you provide a test or steps to replicate.

What ORFS design can I use to create a grt failure and how do I induce it?

oharboe, 2024-11-26:
> @oharboe the list shows all the markers stored in the database, so it's not clear to me how this list contains markers from a different design. Can you provide a test or steps to replicate.

What does it mean that the "marker is stored in the database"? Is there a reference to a file? Is the .rpt file on disk superfluous?

gadfort, 2024-11-26:
@oharboe the rpt on disk is not needed. It's just a copy of what is already in the database.

oharboe, 2024-11-26:
> @oharboe the rpt on disk is not needed. It's just a copy of what is already in the database.

I see... Is this the case for DRC errors from detailed routing too? Are there other sources of such DRC errors in the database?

Yes, in that case it is more of a mystery what is going on.

I can try to create a public test case, but then I need to know which ORFS design that I can induce a grt failure into and how to introduce it.

gadfort, 2024-11-26:
detailed routing also generates markers in the database l, so loading the report in is not needed. PDN, psm, grt, drt can all generate markers.

closed 2024-11-26 by oharboe

eder-matheus, 2024-11-26:
> I can try to create a public test case, but then I need to know which ORFS design that I can induce a grt failure into and how to introduce it.

@oharboe You can indice a congestion error in `nangate45/gcd` by changing the `platforms/nangate45/fastroute.tcl` file and increasing the adjustments to 0.9. It will fail with congestion and the markers will be stored in odb.

oharboe, 2024-11-26:
Thanks!

With designs/gcd/asap7/config.mk, I can reproduce grt failures without modifying any files.

@gadfort I was not able to reproduce using this procedure, so keeping this closed until I can reproduce the problem. Assume PEBKAC meanwhile.

```
make DESIGN_CONFIG=designs/asap7/gcd/config.mk ROUTING_LAYER_ADJUSTMENT=0.9 do-grt gui_5-1_grt-failed.odb
```

```
make DESIGN_CONFIG=designs/asap7/gcd/config.mk gui_5_1_grt-failed.odb
```

Here I can see congestion failures in the dropdown.

If I then run:

```
make DESIGN_CONFIG=designs/asap7/gcd/config.mk do-grt gui_5_1_grt.odb
```

No failures in grt dropdown, as expected.