The-OpenROAD-Project/OpenROAD #6267: manual macro place leads to drt failure

oharboe, 2024-11-29:
### Describe the bug

To reproduce with a full run:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk RTLMP_FLOW=0
```

To reproduce macro placement problem & the detailed routing problems, but not the intervening steps:

untar and run [detail-route-after-manual-place.tar.gz](https://github.com/user-attachments/files/17956203/detail-route-after-manual-place.tar.gz)

```
[deleted]
[WARNING DRT-0418] Term ces_6_3/io_ins_right[15] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_ins_right[26] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_ins_right[28] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_ins_right[38] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_ins_right[42] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_ins_right[7] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbIns_1 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbIns_2 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbIns_3 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbOuts_1 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbOuts_2 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_lsbOuts_3 has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[0] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[10] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[31] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[37] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[4] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_left[5] has no pins on routing grid
[WARNING DRT-0418] Term ces_6_3/io_outs_right[0] has no pins on routing grid
[WARNING DRT-0418] message limit (1000) reached. This message will no longer print.
[deleted]
[INFO DRT-0165] Start pin access.
[ERROR DRT-0073] No access point for ces_3_0/io_ins_left[0].
[ERROR DRT-0073] No access point for ces_1_0/io_ins_left[0].
[INFO DRT-0076]   Complete 1000 pins.
[ERROR DRT-0073] No access point for ces_6_0/io_ins_left[0].
[INFO DRT-0076]   Complete 2000 pins.
[ERROR DRT-0073] No access point for ces_4_0/io_ins_left[0].
Error: detail_route.tcl, 47 DRT-0073
```

### Expected Behavior

manual placement should adjust macro positions such that pins are on routing grid and detailed route works

### Environment

```shell
OpenROAD v2.0-17372-gfa1888e5f
```

### To Reproduce

See above

rovinski, 2024-11-29:
If manual macro placement gets built-in snapping added, there should be an option to turn off snapping in case the designer absolutely needs to place the macro at a specific location.

maliberty, 2024-11-30:
There already is snapping code in MacroPlacer2::placeMacro
```
  if (!orientation.isRightAngleRotation()) {
    Snapper snapper(logger_, inst);
    snapper.snapMacro();
```

Please provide a test case at the point at which you do manual placement.

AcKoucher, 2024-12-02:
@maliberty @rovinski The only case in which we don't snap automatically when doing manual placement is when the macro is being placed with a right angle rotation e.g, R90. Apart from this situation, we always snap.

joaomai, 2024-12-10:
@oharboe 
It looks like all the pins that are unaligned with the track-grid are in M2. The interval between tracks in asap7's M2 layer isn't always the same, which causes problems when snapping the macros' pins with our current snapping mechanism.

Is it imperative to use this layer? When snapping for pins with the same direction in multiple layers was implemented, M2 was not used, so this problem didn't show up.

oharboe, 2024-12-10:
@joaomai If I have clear actionable errors, I can modify my design accordingly.

I dont understand what OpenROAD wants me to do here.

maliberty, 2024-12-14:
I imagine there exists so some snapping that would be m2 and the other layers all on-grid.  Is it an issue with no searching far enough to find such a snapping?

maliberty, 2024-12-14:
Minor note - if the net is connect by abutment then this check is unnecessary.  It should only warn about the pins on the edges of the array.

oharboe, 2024-12-14:
Thanks for the update. My understanding of these matters isnt sophisticated enough to offer any comments.

maliberty, 2024-12-15:
Note the test case doesn't include ${FLOW_HOME}/designs/asap7/mock-array/macro-placement.tcl.  Is it the same as in master?

oharboe, 2024-12-15:
> Note the test case doesn't include ${FLOW_HOME}/designs/asap7/mock-array/macro-placement.tcl. 

Ah, it is missing from ADDITIONAL_FILES in designs/ asap7/mock-array/config.mk, so make issue did t include it.

> Is it the same as in master?

Yes, you can also reproduce without archive running the following on master:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk RTLMP_FLOW=0
```

maliberty, 2024-12-15:
@oharboe please give it a try with head and my quick solution.

oharboe, 2024-12-18:
@maliberty Still fails with:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk RTLMP_FLOW=0 do-floorplan final
[deleted]
OpenROAD v2.0-17656-gc09de2745
[deleted]
[INFO DRT-0165] Start pin access.
[ERROR DRT-0073] No access point for ces_7_0/io_ins_left[11].
[ERROR DRT-0073] No access point for ces_1_0/io_ins_left[11].
[ERROR DRT-0073] No access point for ces_2_0/io_ins_left[11].
[INFO DRT-0076]   Complete 1000 pins.
Error: detail_route.tcl, 47 DRT-0073
Command exited with non-zero status 1
Elapsed time: 0:06.18[h:]min:sec. CPU time: user 8.10 sys 0.31 (136%). Peak memory: 1019004KB.
make[1]: *** [Makefile:769: do-5_2_route] Error 1
make: *** [Makefile:767: results/asap7/mock-array/base/5_2_route.odb] Error 2
```

AcKoucher, 2024-12-18:
@oharboe I'm trying to take a look at this but it doesn't seem to be a macro-placement.tcl in the test case tar.gz.

```
OpenROAD v2.0-17656-gc09de2745 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
source /home/arthur/Downloads/detail-route-after-manual-place/designs/asap7/mock-array/macro-placement.tcl
[ERROR STA-0340] cannot open '/home/arthur/Downloads/detail-route-after-manual-place/designs/asap7/mock-array/macro-placement.tcl'.
Error: macro_place_util.tcl, 74 STA-0340

oharboe, 2024-12-18:
@AcKoucher Try [macro-place-problem.tar.gz](https://github.com/user-attachments/files/18184383/macro-place-problem.tar.gz)

AcKoucher, 2024-12-18:
@oharboe Thanks.

If I run the new test case with master it looks like the pins in m2 get correctly snapped:
_ces_6_3/io_ins_right[15]_
[image]

Same pin in 2_3_floorplan_macro.odb of the previous test case:
[image]

Are you sure that with master we're still having the same problem in DRT? I checked the pins of the other layer and everything looks properly aligned.

AcKoucher, 2024-12-18:
@oharboe Thanks for the new test case.

@maliberty It looks like the quick fix makes some pins slightly unaligned.
Row 7:
[image]

Row 1:
[image]

@osamahammad21 MPL needs fixing here, as these pins should be aligned with the track-grid. However, would it be the case that DRT should be able to handle situations such as these shown in the images above?

oharboe, 2024-12-18:
@osamahammad21 @AcKoucher For final verification, run `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk RTLMP_FLOW=0 nuke final` to verify that macro placement and detailed routing works.

It is probably easier, less confusing and faster for you to run this from scratch locally than to understand a test case cobbled together by me using `make issue` as this failure mode is outside the trodden path: the test case breaks in floorplan, but the problem can only be seen in detailed routing.

maliberty, 2024-12-18:
It is a problem as m2 & m4 both have LEF58_RIGHTWAYONGRIDONLY so such pins are unusable.

maliberty, 2024-12-18:
I think you can call pin_access to get the message though I'm not sure if the unplaced std cells will be a problem.

oharboe, 2024-12-18:
> I think you can call pin_access to get the message though I'm not sure if the unplaced std cells will be a problem.

So it might lead to false positive error messages, if I understand...

AcKoucher, 2024-12-18:
> @AcKoucher Would it be an idea to move the DRT error message as a final step in macro placement? That way there would be a single step that fails quickly and the cause and failure is in the same step.

Generally I'm not sure it makes sense. But perhaps a check for this specific situation.

@maliberty Would it be reasonable throwing an error if we see unaligned pins on LEF58_RIGHTWAYONGRIDONLY layers?

maliberty, 2024-12-18:
I only suggested it for debugging.

joaomai, 2025-01-14:
@oharboe 
PR #3042 was merged and fixes the problem when running `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk RTLMP_FLOW=0 nuke final`. I'm currently working in a more definitive solution in relation to multi-pattern track-grid.

AcKoucher, 2025-05-01:
Definitive approach made in #6360.

closed 2025-05-01 by AcKoucher