The-OpenROAD-Project/OpenROAD #6268: [drt] backtrace data structure problem

beantowel, 2024-11-29:
### Describe the bug

[image]

I suppose `getPrevAstarNodeDir` and `setPrevAstarNodeDir` in `FlexGridGraph_maze.cpp` were used as the backtrace data structure for the maze router (Astar). The algoritm is basically a BFS? search to find lowests cost path from src to dst. But in the situation showed by the picture, it may fails to find the purple path with less bends (which means lower cost, although not supported now, but considered in `getEstCost`). 

It's because the backtrace data may be written by the red path first, and there is no chance for the purple path to proceed from the  `mid` point. The problem is: in this special BFS, we can reach `mid` point twice from different path with the **same cost** but different direction, while the direction matters for the bend count of the latter path.

I suggest we change the signature to `getPrevAstarNodeDir(idx, out)` and `setPrevAstarNodeDir(x, y, z, out, in)` to support more backtrace chances, which means while backtracing, we look for the previous direction(in) taking the current direction(out) into count.

### Expected Behavior

as mentioned before, we shall always find the path with lower bends.

### Environment

```shell
Linux 3.10.0
cmake 3.27.2
CXX compiler GNU 9.5.0
...
```

### To Reproduce

i'm sorry for using a custom script and gds file to test the drt, so it's hard to reproduce. But i've tried to trace the root cause.

maliberty, 2024-11-29:
Its not a bfs, its A*

rovinski, 2024-11-29:
This would definitely require a test case to demonstrate this behavior, because the router already takes these things into account. Assuming the blue path and the red path represent two paths on the same layer, the router has a cost it adds for turns. Once the two separate paths reached the mid point, the blue path would presumably have the lower cost due to the lower turn count. It would only become confusing if there were 3 paths that entered the mid point, because the router currently has an expansion limitation of 2 for any given point.

I'm also pretty sure that turn costs are already accounted in the estimated cost, which means that if there are no obstructions, the router wouldn't even create the red path. It would at minimum route straight up until it is on the same y-coordinate as DST and then route over. In this case the paths would have the same cost.

beantowel, 2024-11-30:
@rovinski the SRC point in this case was a "border" pin on the routing Box edge, its direction was UNKNOWN, so the two paths do have the same est cost at the mid point (which I just printed using the logger)

osamahammad21, 2024-12-03:
@beantowel Certainly the backtrace buffer limitation does affect the final routing solution and may lead missing the optimal path and end up with a sub-optimal path. However, I don't think this would happen in this case. First of all, we do account for the bend count in the estimated cost: https://github.com/The-OpenROAD-Project/OpenROAD/blob/b3acd86f12b561ee57754be3712cd12e3d671a7f/src/drt/src/dr/FlexGridGraph_maze.cpp#L256-L263
Assuming that both paths have the exact same path cost, they would still have different total cost value (total = path_cost + est_cost). This makes the queue entry from the blue path more preferable (less cost) in the A* queue. Meaning that theoritcally, the blue path would reach the **mid** point before the red path. Moreover, this still would not stop both paths from proceeding after reaching the **mid** point. The backtrace buffer's size is 3 nodes. Meaning each wavefront holds up to the latest 3 previous expansion directions. So the decision whether to take the red path or the blue path does not happen at the **mid** point, but happens 3 nodes later.
[image]
Still the blue path would reach the **decision** point before the red path because it has less path cost as we also account for the bend count in the path cost:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/b3acd86f12b561ee57754be3712cd12e3d671a7f/src/drt/src/dr/FlexGridGraph_maze.cpp#L393-L401
In conclusion, I agree that the backtrace buffer limits the solution space. However, it's a necessary compromise as finding the optimal path without such constraints would be computationally unaffordable.
If you have a test case, I’d be happy to take a look to identify the real issue.

osamahammad21, 2024-12-17:
I am gonna close this issue since there are no updates for 2 weeks. @beantowel feel free to reopen if you have any updates.

closed 2024-12-17 by osamahammad21