The-OpenROAD-Project/OpenROAD #6287: RUDY routing congestion heatmap correlation to fastroute congestion

gudeh, 2024-12-03:
### Description

We currently have a slight mismatch between RUDY's congestion estimated heatmap and the more exact one provided by fastroute. RUDY can fairly well indicate the correct regions with high congestion, matching with the same regions of fastroute. Although, the intensity of RUDY values seems too low.

Here is ng45/bp, RUDY is by the right:
[image]

Modifications on RUDY congestion values should have implications on gpl routability driven mode.

stefanottili, 2024-12-05:
While somebody will work on RUDY, https://github.com/The-OpenROAD-Project/OpenROAD/issues/5557 is related.
Differences between gpl and RUDY routing resources, wrong resource calculation on top/right edges that have slightly different gcell sizes and the fact that RUDY displays the map wrong for floorpans where left/bottom is not 0,0.

gudeh, 2025-02-04:
fixed with PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/6590

closed 2025-02-04 by gudeh

stefanottili, 2025-02-04:
running OpenROAD v2.0-18772-gc05b0dd78 (Feb 4 2025)
revisiting https://github.com/The-OpenROAD-Project/OpenROAD/issues/5548 ICCAD04/RISC2 first_quadrant 32k stdcells
running without any adjustments "out of the box"
RUDY and global route still differ quite a bit.
The differences is smaller when using the adjustments.
```
#set_routing_layers -signal metal1-metal5
#set_global_routing_layer_adjustment * 0.3
#global_placement -density 0.8
```
RUDY
[image]
global route
[image]

gudeh, 2025-02-04:
Thanks for sharing @stefanottili, how many layers does the PDK used for this design has?