The-OpenROAD-Project/OpenROAD #6290: Pledge(2) like utility for odb

QuantamHD, 2024-12-03:
### Description

Add a utility to OpenROAD that lets pass authors pledge away certain operations during their code execution. If a pass performs an operation that it pledged not to odb would throw an error. Which would help prevent bugs like https://github.com/The-OpenROAD-Project/OpenROAD/pull/6289 

### Suggested Solution

These pledges are similar in nature to the FreeBSD pledge(2) syscall that lets process restrict the syscalls they can make as a security feature.

Ideally we would have some built-in mechanism to odb to pledge away write/delete operations on certain objects. Potentially, using RAII to undo them automatically.

maliberty, 2024-12-03:
Wouldn't a temporary dont-touch be equivalent?  With only the one example I'm not sure what operations/semantics need pledging.

QuantamHD, 2024-12-04:
But you would have to set that on each net of those types. This would essentially be a global precheck.

maliberty, 2024-12-04:
Would you illustrate with some examples where this works be helpful?