The-OpenROAD-Project/OpenROAD #6297: Build fail on latest OR version

hpretl, 2024-12-04:
### Describe the bug

I try to build latest OR on Ubunutu 22.04 LTS, and I get the following build error:
```
[ 92%] Building CXX object src/drt/CMakeFiles/drt.dir/src/dr/FlexGridGraph_maze.cpp.o
In file included from /usr/include/spdlog/fmt/fmt.h:25,
                 from /usr/include/spdlog/common.h:36,
                 from /usr/include/spdlog/details/os.h:6,
                 from /tmp/OpenROAD/src/utl/include/utl/Logger.h:51,
                 from /tmp/OpenROAD/src/odb/include/odb/isotropy.h:50,
                 from /tmp/OpenROAD/src/odb/include/odb/geom.h:40,
                 from /tmp/OpenROAD/src/drt/include/triton_route/TritonRoute.h:42,
                 from /tmp/OpenROAD/src/drt/src/dr/FlexDR.h:31,
                 from /tmp/OpenROAD/src/drt/src/dr/FlexGridGraph_maze.cpp:29:
/usr/include/fmt/core.h: In instantiation of ‘constexpr fmt::v8::detail::value<Context> fmt::v8::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v8::basic_format_context<fmt::v8::appender, char>; fmt::v8::detail::type <anonymous> = fmt::v8::detail::type::custom_type; T = drt::frDirEnum&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]’:
/usr/include/fmt/core.h:1855:77:   required from ‘constexpr fmt::v8::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, odb::Point&, unsigned int&, unsigned int&, drt::frDirEnum&, unsigned int&, int&, int&}; Context = fmt::v8::basic_format_context<fmt::v8::appender, char>; Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, odb::Point, unsigned int, unsigned int, drt::frDirEnum, unsigned int, int, int}]’
/usr/include/fmt/core.h:1872:38:   required from ‘constexpr fmt::v8::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<Args>::type>::type ...> fmt::v8::make_format_args(Args&& ...) [with Context = fmt::v8::basic_format_context<fmt::v8::appender, char>; Args = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, odb::Point&, unsigned int&, unsigned int&, drt::frDirEnum&, unsigned int&, int&, int&}]’
/usr/include/fmt/core.h:3119:44:   required from ‘std::string fmt::v8::format(fmt::v8::format_string<T ...>, T&& ...) [with T = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int, odb::Point&, unsigned int, unsigned int, drt::frDirEnum, unsigned int, int&, int&}; std::string = std::__cxx11::basic_string<char>; fmt::v8::format_string<T ...> = fmt::v8::basic_format_string<char, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int, odb::Point&, unsigned int, unsigned int, drt::frDirEnum, unsigned int, int&, int&>]’
/tmp/OpenROAD/src/drt/src/dr/FlexGridGraph_maze.cpp:53:28:   required from here
/usr/include/fmt/core.h:1728:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1728 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1728:7: note: ‘formattable’ evaluates to false
make[2]: *** [src/drt/CMakeFiles/drt.dir/build.make:310: src/drt/CMakeFiles/drt.dir/src/dr/FlexGridGraph_maze.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:6135: src/drt/CMakeFiles/drt.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

### Expected Behavior

Build should be clean.

### Environment

```shell
I am using stock spdlog with this version:
libspdlog-dev/jammy,now 1:1.9.2+ds-0.2 arm64 [installed]
```

### To Reproduce

Do a standard build on Ubuntu 22.04 LTS.

### Relevant log output

```shell
See above.
```

maliberty, 2024-12-04:
Please use the dependency installer to get the supported version.

hpretl, 2024-12-04:
`spdlog` version 1.8.1 (which works fine, and I build it locally now for OR) is ca. 4 years old. Until ca. 8 weeks back, 1.9.x worked fine with OR, so maybe an upgrade to a more modern version of `spdlog` would be worth the effort?

stefanottili, 2024-12-04:
I'm a complete newbie to fmt and c++, but the error message states that currGrid.getLastDir() is an enum and it seems that fmt just doesn't know what of print for that. The other two fmt::format statements compile just fine.
Commenting out currGrid.GetLatDir() makes the compile go thru on MacOS M1 that had the same issue as Ubuntu 22.
```
 53   dump_file_ << fmt::format(
 54       "{} {} pt {} cost {} pathCost {} lastDir " // "{} "
 55       "estCost {} gridX {} gridY ",
 56       "{} ",
 57       keyword,
 58       currGrid.z(),
 59       pt,
 60       currGrid.getCost(),
 61       currGrid.getPathCost(),
 62 //      currGrid.getLastDir(),
 63       currGrid.getCost() - currGrid.getPathCost(),
 64       gridX,
 65       gridY);
```

maliberty, 2024-12-06:
Its time https://github.com/The-OpenROAD-Project/OpenROAD/pull/6315

maliberty, 2024-12-06:
Merged

closed 2024-12-06 by maliberty