The-OpenROAD-Project/OpenROAD #6316: grt::have_routes should be silent when elements are not placed

oharboe, 2024-12-06:
### Description

```
make gui_floorplan
```

Results in:

```
>>> grt::have_routes
[ERROR GRT-0010] Instance ctrl.state.out\[0\]$_DFF_P_ is not placed.
[ERROR GUI-0070] GRT-0010
```

### Suggested Solution

silently return 0

### Additional Context

Useful in scripting

maliberty, 2024-12-06:
Didn't you already make a PR for this?

oharboe, 2024-12-06:
> Didn't you already make a PR for this?

This is the next bug in the same vein

maliberty, 2024-12-07:
Are you planning to fix this?  I assigned the last one only to have you fix it.

oharboe, 2024-12-07:
> Are you planning to fix this?  I assigned the last one only to have you fix it.

No, I have a workaround. The last one turned out to be trivial to fix and it was blocking me.

This one is more about consistency, hopefully trivial too though.

closed 2024-12-11 by tspyrou