The-OpenROAD-Project/OpenROAD #6323: Improve error message [ERROR GPL-0305] RePlAce diverged at newStepLength.

oharboe, 2024-12-07:
### Description

I wanted to have a look at what happens when there are long wires to distribute a signal:

[image]

However, I get a mysterious error message in global placement.

I think global placement wants me to tweak some variable or other, but the error message doesn't give me a clue what to fix.

untar and run [global_place_skip_io_Bus_asap7_766_2024-12-07_15-05.tar.gz](https://github.com/user-attachments/files/18048674/global_place_skip_io_Bus_asap7_766_2024-12-07_15-05.tar.gz)

```
OpenROAD v2.0-17210-g8300b5293 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
global_placement -skip_io -density 0.40 -pad_left 0 -pad_right 0
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: (  0.054  0.270 ) um
[INFO GPL-0004] CoreBBox: (  2.052  2.160 ) ( 763.992 17.820 ) um
[INFO GPL-0006] NumInstances:              1035
[INFO GPL-0007] NumPlaceInstances:           19
[INFO GPL-0008] NumFixedInstances:         1016
[INFO GPL-0009] NumDummyInstances:            0
[INFO GPL-0010] NumNets:                     22
[INFO GPL-0011] NumPins:                     40
[INFO GPL-0012] DieBBox:  (  0.000  0.000 ) ( 766.000 20.000 ) um
[INFO GPL-0013] CoreBBox: (  2.052  2.160 ) ( 763.992 17.820 ) um
[INFO GPL-0016] CoreArea:             11931.980 um^2
[INFO GPL-0017] NonPlaceInstsArea:       29.627 um^2
[INFO GPL-0018] PlaceInstsArea:           1.823 um^2
[INFO GPL-0019] Util:                     0.015 %
[INFO GPL-0020] StdInstsArea:             1.823 um^2
[INFO GPL-0021] MacroInstsArea:           0.000 um^2
[INFO GPL-0031] FillerInit:NumGCells:     23710
[INFO GPL-0032] FillerInit:NumGNets:         22
[INFO GPL-0033] FillerInit:NumGPins:         40
[INFO GPL-0023] TargetDensity:            0.400
[INFO GPL-0024] AvrgPlaceInstArea:        0.096 um^2
[INFO GPL-0025] IdealBinArea:             0.240 um^2
[INFO GPL-0026] IdealBinCnt:              49757
[INFO GPL-0027] TotalBinArea:         11931.980 um^2
[INFO GPL-0028] BinCnt:      1024     32
[INFO GPL-0029] BinSize: (  0.745  0.490 )
[INFO GPL-0030] NumBins: 32768
[NesterovSolve] Iter:    1 overflow: 0.934 HPWL: 5131
[NesterovSolve] Iter:   10 overflow: 1.190 HPWL: 1460
[NesterovSolve] Iter:   20 overflow: 1.158 HPWL: 1043
[NesterovSolve] Iter:   30 overflow: 1.152 HPWL: 977
[NesterovSolve] Iter:   40 overflow: 1.156 HPWL: 924
[deleted]
[NesterovSolve] Iter: 2200 overflow: 0.670 HPWL: 114440
[NesterovSolve] Iter: 2210 overflow: 0.670 HPWL: 113127
[NesterovSolve] Iter: 2220 overflow: 0.670 HPWL: 110586
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: global_place_skip_io.tcl, 21 GPL-0305
openroad> 
```

### Suggested Solution

Make the global placement work or make the error message actionable.

oharboe, 2024-12-07:
Not related to this issue as such, but I reduced the width and I got a result.

gudeh, 2024-12-09:
Hi @oharboe, this is where the error report comes from:
[image]

I would say this is more then a divergence, it is an actual crash because newStepLength was calculated as NAN or infinity. I will investigate why. Also, I think changing the wording for this report is necessary, since it is an actual crash.

gudeh, 2024-12-09:
@oharboe would you happen to know why I was not able to run your issue with debug mode? Furthermore, I noticed that gpl never converged to the 0.10 overflow, I want to investigate that.

oharboe, 2024-12-09:
> @oharboe would you happen to know why I was not able to run your issue with debug mode?

Don't know. Sorry.

gudeh, 2024-12-10:
Using a lower density such as 0.10, gpl converges.

I believe we get stuck in 0.67 overflow when using 0.40 density, as provided by Oyvind, because of the bin area scaling in line https://github.com/The-OpenROAD-Project/OpenROAD/blob/5daa1de769fbe1ae6451325e760b79ca13987041/src/gpl/src/nesterovBase.cpp#L892. However, it would be good to confirm 

I wonder how we could avoid this issue while still making it work with a 0.40 density. My understanding is that a lower actual density than the target density should still be acceptable.

titan73, 2024-12-12:
I encountered the same message https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6338. It takes me some times that the density was too high as the message is not very straigtforwad.
That would be good to have a better message saying which metric is overflowing. Something like "Density is too high (120%), aborting" for example.

gudeh, 2025-01-27:
Hey guys, just some updates on this issue. Thanks to this "unusual" design provided by @oharboe, @LucasYuki was able to detect that we sometimes might have negative bin sizes on the border of the bin grid. A solution was already implemented and we are just fixing some CI errors to have the fix merged.

We also noticed that TAP cells are sometimes ignored by gpl. For example, this design we completely ignore the existence of TAP cells, [there is an if condition in the code we still need to understand why it exists](https://github.com/The-OpenROAD-Project/OpenROAD/blob/df581bed847c639c4cd4fe5041a26db42631fd8a/src/gpl/src/placerBase.cpp#L825).

gudeh, 2025-01-27:
Concerning the error message. We have other divergences in gpl, but this case specifically is when the step length is calculated as NaN or infinity. Taking into account @QuantamHD 's explanation on the link sent by @titan73, I thought of something like:

`[ERROR GPL-0305] RePlAce diverged during gradient descent calculation, resulting in an invalid step length (Inf or NaN). This is often caused by numerical instability or extreme placement density. Consider reducing placement density to potentially resolve the issue.`

gudeh, 2025-05-05:
Hi @oharboe the design should run successfully with the changes from https://github.com/The-OpenROAD-Project/OpenROAD/pull/6530#event-17522018082, where we fixed the negative bin sizes bug. We also modified the error message based on the discussion.

closed 2025-05-05 by gudeh