The-OpenROAD-Project/OpenROAD #6339: Paths does not have the same hierarchy separator ("\/" or "/")

titan73, 2024-12-11:
### Describe the bug

Messages containing paths with "\/" hierarchy separator (Note "/" is used to add the pin):
```
[ERROR DRT-0073] No access point for clkbuf_0_I_digital_top\/alwayson_top_i\/s_clk4M_dft/A.
[INFO CTS-0027] Generating H-Tree topology for net I_digital_top\/nvm_control_top_0\/u_nvm_mmu_softip_top\/u_nvm_ctrl\/c_nvm_wrapper\/n608.
[INFO CTS-0098] Clock net "I_digital_top\/computing_top_i\/average_inst\/net14332"
[WARNING CTS-0041] Net "I_digital_top\/alwayson_top_i\/reg_top_i\/s_power_uv_l_reg_clk" has 1 sinks. Skipping...
```

Messages containing paths with "/" hierarchy separator:
```
[ERROR DRT-0073] No access point for I_digital_top/alwayson_top_i/reg_top_i/U82/A.
INFO CTS-0007] Net "I_digital_top/alwayson_top_i/clkgen_i/int_clk31K" found for clock "clock_osc_4m_div4_div32".
[INFO CTS-0010]  Clock net "I_digital_top/nvm_control_top_0/u_nvm_mmu_softip_top/u_nvm_ctrl/c_nvm_wrapper/s_clk_pckdiv2" has 4 sinks.
[WARNING DRT-0120] Large net I_digital_top/nvm_control_top_0/u_cyclic_crc_check/n311 has 114 pins which may impact routing performance. Consider optimization.
[WARNING STA-0363] pin 'I_digital_top/nvm_control_top_0/u_nvm_mmu_softip_top/u_mux_nvm_remap_rst_n/Z' not found.
```

Even wierder, [ERROR DRT-0073] can have both separators???

Is there supposed to be a standard separator?

### Expected Behavior

Same separator

### Environment

```shell
Latest git
```

### To Reproduce

Whatever test case.

titan73, 2024-12-13:
Hierarchy separator '/' is escaped for clock nets in src/cts/src/Clock.cpp:
```
Clock::Clock(const std::string& netName,
             const std::string& clockPin,
             const std::string& sdcClockName,
             int clockPinX,
             int clockPinY)
    : clockPin_(clockPin),
      sdcClockName_(sdcClockName),
      clockPinX_(clockPinX),
      clockPinY_(clockPinY)
{
  // Hierarchy delimiters in the net name must be escape.  We use
  // the name to construct buffer names later and the delimiters
  // will confuse downstream tools like read_spef.
  netName_ = sta::escapeChars(netName.c_str(), '/', '\0', '\\');
}
```
Disabled escaping makes all messages coherents.
Is escaping clock nets is this still relevant? Signal nets are not escaped and there is no issue apparently. Do I miss something?

maliberty, 2024-12-18:
How does changing cts fix drt?  Names in EDA are generally awful.  I would guess you have escaped verilog names in your netlist.  Without a test case it is hard to say.

titan73, 2024-12-18:
Not sure I understand your question about "cts fix drt".
Here is a simple test case:
[escaped_clock.zip](https://github.com/user-attachments/files/18186085/escaped_clock.zip)

The net became escaped once the the Clock object is created and used:
```
[INFO CTS-0010]  Clock net "mytop/myclk" has 100 sinks.
[INFO CTS-0008] TritonCTS found 1 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 0 placement blockages have been identified.
[INFO CTS-0201] 0 placed hard macros will be treated like blockages.
[INFO CTS-0027] Generating H-Tree topology for net mytop\/myclk.
```

maliberty, 2024-12-18:
`Even wierder, [ERROR DRT-0073] can have both separators???`
why would changing CTS fix that?

titan73, 2024-12-18:
Test case was done with OpenROAD v2.0-17609-gced2b70f4.

titan73, 2024-12-18:
> `Even wierder, [ERROR DRT-0073] can have both separators???` why would changing CTS fix that?

It seems CTS creates nets or instances whose names comes from the Clock object.
There are only 3 in my case (not the test case):
```
[ERROR DRT-0073] No access point for clkbuf_4_4_0_I_digital_top\/nvm_control_top_0\/u_nvm_mmu_softip_top\/u_nvm_ctrl\/c_nvm_wrapper\/net14310/A.
[ERROR DRT-0073] No access point for clkbuf_2_0__f_I_digital_top\/s_clk_nvm/A.
[ERROR DRT-0073] No access point for clkbuf_0_I_digital_top\/alwayson_top_i\/s_clk4M_dft/A.
```

First 2 are from the following code in HTreeBuilder.cpp method void HTreeBuilder::createClockSubNets():
```
    SegmentBuilder builder("clkbuf_1_" + std::to_string(idx) + "_",
                           "clknet_1_" + std::to_string(idx) + "_",
                           legalCenter,  // center may have moved, don't use
                                         // sinkRegion_.getCenter()
                           legalBranchPoint,
                           topLevelTopology.getWireSegments(),
                           clock_,
                           rootClockSubNet,
                           *techChar_,
                           wireSegmentUnit_,
                           this);
```
In DRT, the log message just prints the name of objects:
```
      logger_->error(DRT,
                     73,
                     "No access point for {}/{}.",
                     inst_term->getInst()->getName(),
                     inst_term->getTerm()->getName());
```
Actually not weird at all. If CTS creates objects with escaped names, DRT just prints them as is.

maliberty, 2024-12-18:
The comment seems to explain the need
```
  // Hierarchy delimiters in the net name must be escape.  We use
  // the name to construct buffer names later and the delimiters
  // will confuse downstream tools like read_spef.
```
Does the extra escaping matter?  It isn't a question of having different hierarchy separators.

titan73, 2025-01-20:
Well bare / is easier and prettier to read. I've never seen escaped / so far with proprietary tools I used and the designs I worked on.
So that's why I'm wondering why escaping is needed. Is it OpenSTA that needs it and why?
Also why only clock nets names needs to be escaped and OpenSTA does not struggle on the other nets names that are not escaped.

maliberty, 2025-01-21:
The example you provided only covers CTS.  In that case we escape the names in order to generate new names without conflicts.  It would be more interesting to see the drt message

titan73, 2025-02-11:
> The example you provided only covers CTS. In that case we escape the names in order to generate new names without conflicts. It would be more interesting to see the drt message

titan73, 2025-02-11:
> The example you provided only covers CTS. In that case we escape the names in order to generate new names without conflicts.

In the example, the netlist has 2 clock nets:
```
clk
mytop/myclk
```
CTS creates new unique clock nets by using a prefix with numbers.
```
clknet_0_mytop\/myclk
clknet_3_0__leaf_mytop\/myclk
clknet_3_1__leaf_mytop\/myclk
clknet_3_2__leaf_mytop\/myclk
clknet_3_3__leaf_mytop\/myclk
clknet_3_4__leaf_mytop\/myclk
clknet_3_5__leaf_mytop\/myclk
clknet_3_6__leaf_mytop\/myclk
clknet_3_7__leaf_mytop\/myclk
```
These names are already unique. How escaping / helps here?

> It would be more interesting to see the drt message

Are you referring to the "[ERROR DRT-0073] No access point ...." message?
If so, I already filed an issue https://github.com/The-OpenROAD-Project/OpenROAD/issues/5975.
It seems specific to one of my company techno.