The-OpenROAD-Project/OpenROAD #6345: Parallelization of `FastRouteCore::mazeRouteMSMD`

eszpotanski, 2024-12-12:
### Description

A major part of global routing is spent on fixing overflows in `FastRouteCore::mazeRouteMSMD`. It iterates over Steiner trees representing parts of the routing, which are mostly disjoint. It should be possible to parallelize this method.

### Suggested Solution

We're currently trying to parallelize the `FastRouteCore::mazeRouteMSMD`.

stefanottili, 2024-12-13:
Is this effort inspired by some of the ispd24 contest approaches or do you have something else in mind ?
https://liangrj2014.github.io/ISPD24_contest/
The largest of the ispd24 test cases OpenRoad is currently able to grt/drt in 64GB is the 715k stdcell bsg_chip.def.gz

eszpotanski, 2024-12-16:
No, I though about adjusting existing method to make it work in parallel on CPU.

I checked out the contest and it was mainly focused on GPU acceleration and ML solutions - top submissions are vaguely described in final presentation, but I found a few papers created by the top teams; I'll check whether there is something that can be used.

stefanottili, 2024-12-16:
One of the top three didn’t use the gpu at all, so threading can help a lot with regard to runtime.
Could you point me to the papers you found ?

maliberty, 2024-12-16:
It would be good to describe your approach once you clarify it.  [InstantGR](https://shijulin.github.io/files/1239_Final_Manuscript.pdf) does work with the gpu.

eszpotanski, 2024-12-17:
It seems that papers don't describe work from the contest, as they usually compare results with top three teams. But they are created by the same people, so I guess papers describe improved solutions:
* 1st place team - [paper](https://yibolin.com/publications/papers/ROUTE_ICCAD2024_Zhao.pdf)
* 2nd place team (already mentioned InstantGR) - [paper](https://shijulin.github.io/files/1239_Final_Manuscript.pdf), [code](https://github.com/cuhk-eda/InstantGR/tree/main)

Both approaches use GPU.

Other papers based on ISPD24:
* [paper](https://dl.acm.org/doi/pdf/10.1145/3626184.3633324), [code](https://github.com/SNU-suwan/ISPD24/tree/main)