The-OpenROAD-Project/OpenROAD #6352: cts fails to build clock tree when clock is also an output port

gadfort, 2024-12-13:
### Describe the bug

In this design the clock is getting forwarded to an output port, ie. `assign mclk = dco_clk`
This causes openroad to put a buffer between the two ports, which appears to cause CTS to skip clock tree generation all together.

### Expected Behavior

Build a clock tree on `dco_clk`

### Environment

```shell
OpenROAD v2.0-17598-ga008522d8 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

### To Reproduce

https://drive.google.com/file/d/1GijuZkVXgaxFMtEAyJejZCT3I8CQEH9k/view?usp=sharing

```
tar xvf sc_issue_openmsp430_job0_cts.clock_tree_synthesis0_20241213-143445.tar.gz
cd sc_issue_openmsp430_job0_cts.clock_tree_synthesis0_20241213-143445
./run.sh
```

### Relevant log output

```shell
[INFO CTS-0007] Net "dco_clk" found for clock "clk".
[WARNING CTS-0105] Net "dco_clk" already has clock buffer _12043_. Skipping...
[WARNING CTS-0105] Net "smclk" already has clock buffer _12043_. Skipping...
[WARNING CTS-0105] Net "mclk" already has clock buffer _12036_. Skipping...
[WARNING CTS-0105] Net "aclk" already has clock buffer _12029_. Skipping...
[WARNING CTS-0083] No clock nets have been found.
[INFO CTS-0008] TritonCTS found 0 clock nets.
[WARNING CTS-0082] No valid clock nets in the design.
[INFO RSZ-0058] Using max wire length 232um.
```

precisionmoon, 2025-01-17:
As a temporary workaround, can you manually remove the feedthrough buffers before CTS?

gadfort, 2025-01-17:
@precisionmoon as a temporary bypass yes, I think that should be doable.

precisionmoon, 2025-01-17:
@gadfort, thanks for the testcase.  We'll review it to see how we can eliminate the workaround.

gadfort, 2025-04-14:
@maliberty @precisionmoon any update on this?

maliberty, 2025-04-14:
@arthurjolo please update

arthurjolo, 2025-04-14:
Hi Peter, sorry for not having looking into this it was a bit lost in my issues list. It seems that CTS is not checking the source of this buffer in the clock net so it thinks that it is a clock tree buffer and.

I have found a way to prevent this and will create a PR as soon as possible.

closed 2025-04-15 by maliberty