The-OpenROAD-Project/OpenROAD #6385: manual macro placement with snapping to legal coordinates should complain about suspicious macro placements

oharboe, 2024-12-19:
### Description

place_macro will by default snap to legal coordinates, but it does not complain on overlapping macros.

Such an error on the user's part is annoying, because it will work fine all the way up until global routing, which will fail to converge. It can take hours to generate an error message about failure do global routing.

What is worse is that grt will fail in a very unobvious fashion: it can have tens of thousands of DRC errors as grt will stop trying early and what is shown as DRC errors are only the first 10000 horizontal and vertical DRC errors in global routing. These first 10000 DRC errors could have been easily resolved if it wasn't for the overlap, but the overlap DRC errors are not shown in the GUI as the DRC errors are capped at 10000. I was only able to figure this out by doing a deltaDebug.py run on grt + help from @maliberty 

Overlapping macros to me sounds suspicous and if is probably not what the user intended if they asked to have macros placed at legal coordinates.

I mention overlapping macros here as "supicious macro placement", but perhaps there are other things that can also be categorized as "supicious"?

### Suggested Solution

Add a `-literal` option that disables snapping and checks for sensible macro placements.

AcKoucher, 2024-12-19:
@oharboe I don't think we need to include the overlap part in the new flag. Overlapping seems enough "wrong" by itself so it could be the default behavior. The new flag can be only used to avoid snapping.

maliberty, 2024-12-20:
Some people design macros that can be overlapped.  I don't think we should prevent that entirely.

PR #6403 (2024-12-23): mpl2: new flags for individual manual macro placement
Resolve #6385 

Added:

| Flag | Behavior |
| ---------------- | ------------ |
|  -exact  | Enforce exact location set. Default is to snap. |
|  -allow_overlap  | Places macro if even if it overlaps with other macros. Default is to not allow the macro to be place if it generates overlap with other macros. |
closed 2024-12-23 by eder-matheus