The-OpenROAD-Project/OpenROAD #6394: Interaction between macro placement blockages, rsz repair_design, and gpl

gudeh, 2024-12-19:
### Description

Right after a non-virtual timing-driven iteration by gpl, where we keep the resizing and buffers inserted by rsz repair_design we can see instances added on top of macro placement blockages, such as the following image:
[image]

Gpl does not allow for instances on top of macros, so it moves the instances outside of the positions set by rsz, which are further away, potentially undoing rsz's work:
[image]

I wonder if we should worry about this behavior.

maliberty, 2024-12-23:
rsz needs to be blockage aware.  Right now it just uses Steiner trees and placing instances along the tree edges without such consideration.  Its more of an rsz issue than a gpl one.

maliberty, 2024-12-23:
@gudeh are you wanting to look into this or pass it to @precisionmoon ?

gudeh, 2024-12-24:
If rsz needs to be blockage-aware, I think it would make more sense for someone with more experience in rsz to tackle this issue.

Furthermore, I wonder if this behavior has some impact on gpl divergences. Currently, I suspect that the change in area is the cause for gpl diverging.