The-OpenROAD-Project/OpenROAD #6400: Stop search instance "*FF*" from locking up the GUI

oharboe, 2024-12-20:
### Description

For a large design, it is easy to accidentally launch a search that is too broad, this will then lock up the GUI.

[image]

### Suggested Solution

Press "ESC"  to stop search

rovinski, 2024-12-22:
Just to add, many commercial tools throw a popup at ~10k instances asking if you are sure you want to continue.

maliberty, 2024-12-22:
It comes back to the general issue of needing a cancel option on many long running commands.   The same sort as https://github.com/parallaxsw/OpenSTA/issues/111

oharboe, 2024-12-22:
> It comes back to the general issue of needing a cancel option on many long running commands.   The same sort as https://github.com/parallaxsw/OpenSTA/issues/111

The use case here is a bit different: it is used as a beggars heatmap for flip flops, wires, etc.

oharboe, 2024-12-22:
Also, I think there is a performance problem unqiue to instances, I dont recall it locking up on wires.

Perhaps the performance can be fixed and we can forget all about it?

LucasYuki, 2025-01-13:
The problem was with the code to search for instances in the clock tree viewer. Each instance found there was triggering a redraw.
Blocking the redraw when updating the selected instances in the clock tree viewer seems to fix the performance problem.

PR #6514 (2025-01-13): gui: optimizing find in clock tree viewer
Closes #6400.

Changes:
- Optimizes instance search in CTS Viewer by disabling rendering when setting nodes to selected
  - Previously, the CTS Viewer was rendering everything after each node was set to selected
- Changes items name hash table to unordered_map