The-OpenROAD-Project/OpenROAD #6404: Make global routing easier to use, log estimated maximum ROUTING_LAYER_ADJUSTMENT values

oharboe, 2024-12-21:
### Description

If ROUTING_LAYER_ADJUSTMENT is too high, then there's no hope of global routing succeeding and it takes hours to fail.

To my mind, for extremely congested deigns, where ROUTING_LAYER_ADJUSTMENT is much too high and the run is doomed anyway, the job of the global router is to give actionable feedback.

Global route failures are not easy or quick to debug.

Today I have to set up a sweep of ROUTING_LAYER_ADJUSTMENT values (using bazel-orfs), examine results, decide if I need to run deltaDebug.py to whittle down the problem further.

A high ROUTING_LAYER_ADJUSTMENT makes the job easier for detailed routing and improves the quality of results, as I undrestand it. I'm unsure if a high ROUTING_LAYER_ADJUSTMENT has any negative effects.

### Suggested Solution

Some ideas...

- Before starting global routing, make an estimate of maximum ROUTING_LAYER_ADJUSTMENT and log it, something like: "Estimated maximum ROUTING_LAYER_ADJUSTMENT is 0.25".
- The global routing has a lot of logged output, can it be trimmed down so as on to water down user actionable information?

tspyrou, 2024-12-21:
@eder-matheus could the placement based congestion analysis be used to suggest the adjustment numbers?

maliberty, 2024-12-21:
Setting it to a high value can cause unneeded detouring.  Ideally the user wouldn't need to set it at all gpl routability would figure it out.  I don't really know how to estimate it that well or we would just automate it.

oharboe, 2024-12-21:
I see... For now I "estimate" using a value sweep(using bazel-orfs).

If there is nothing to be done, perhaps elaborate the docs a bit?

maliberty, 2024-12-21:
What text do you suggest?

oharboe, 2024-12-22:
How about https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2637 ?