The-OpenROAD-Project/OpenROAD #6410: tcllib (tcl standard library) not usable in interpreter

donn, 2024-12-22:
### Describe the bug

Attempting to `package require lambda` (or anything in tcllib) yields a message about not being able to read the variable `dir`: `Error: pkgIndex.tcl, 11 can't read "dir": no such variable` 

The Tcl wiki claims that this variable should always exist when `pkgIndex.tcl` files are read: https://wiki.tcl-lang.org/page/dir

### Expected Behavior

It'd be pretty darn useful if this worked.

### Environment

```shell
Git commit: 51302eb80b11576a01171d33452c362301d55143
kernel: Linux 6.8.0-51-generic
os: Ubuntu 24.04 LTS (Noble Numbat)
cmake version 3.24.2
-- The CXX compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17776-g51302eb80
-- System name: Linux
-- Compiler: GNU 13.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 13.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 99 lines cut ...]
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.MvPGNhaTbr
```

### To Reproduce

```
./build/src/openroad -exit <(echo "package require lambda") 
```

### Relevant log output

```shell
OpenROAD v2.0-17776-g51302eb80 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
error reading package index file /usr/lib/tcltk/aarch64-linux-gnu/tclreadline2.3.8/pkgIndex.tcl: Error: pkgIndex.tcl, 11 can't read "dir": no such variable
error reading package index file /usr/share/tcltk/tcllib1.21/pkgIndex.tcl: Error: pkgIndex.tcl, 11 can't read "dir": no such variable
error reading package index file /usr/share/tcltk/tcl8.6/http1.0/pkgIndex.tcl: Error: pkgIndex.tcl, 11 can't read "dir": no such variable
error reading package index file /usr/share/tcltk/tcl8.6/opt0.4/pkgIndex.tcl: Error: pkgIndex.tcl, 12 can't read "dir": no such variable
Error: 16, 1 can't find package lambda
```

donn, 2024-12-22:
Related/possible duplicate of #5875

oharboe, 2025-01-09:
> Related/possible duplicate of #5875

Yes, duplicate.

Close this issue?

closed 2025-01-09 by donn