The-OpenROAD-Project/OpenROAD #6411: dft::ScanArchitect not cross-OS deterministic?

donn, 2024-12-22:
### Describe the bug

The following tests pass on Linux but fail on macOS, ostensibly due to the chain somehow returning in a different order:

* dft.scan_architect_clock_mix_sky130.tcl
* dft.scan_architect_no_mix_sky130.tcl
* dft.scan_architect_register_bank_no_clock_mix_sky130.tcl

### Expected Behavior

Tests should not be sensitive to the operating system. This implies the behavior of dft::ScanArchitect is not OS-independent.

### Environment

```shell
Git commit: 51302eb80b11576a01171d33452c362301d55143
kernel: Darwin 24.1.0
os: macOS 15.1
cmake version 3.30.5
CMake Warning (dev) at src/CMakeLists.txt:175 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at src/CMakeLists.txt:267 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Warning (dev) at src/dpl/CMakeLists.txt:41 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/cts/CMakeLists.txt:36 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/drt/CMakeLists.txt:49 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/utl/CMakeLists.txt:38 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/dst/CMakeLists.txt:42 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at src/dft/test/CMakeLists.txt:20 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.
[... 199 lines cut ...]
=====

Test project /Users/donn/efabless/openroad/build-mac
    Start 963: dft.scan_architect_clock_mix_sky130.tcl
1/3 Test #963: dft.scan_architect_clock_mix_sky130.tcl ....................***Failed    0.38 sec
    Start 964: dft.scan_architect_no_mix_sky130.tcl
2/3 Test #964: dft.scan_architect_no_mix_sky130.tcl .......................***Failed    0.38 sec
    Start 965: dft.scan_architect_register_bank_no_clock_mix_sky130.tcl
3/3 Test #965: dft.scan_architect_register_bank_no_clock_mix_sky130.tcl ...***Failed    0.38 sec

0% tests passed, 3 tests failed out of 3

Label Time Summary:
IntegrationTest tcl dft log_compare    =   1.14 sec*proc (3 tests)

Total Test time (real) =   1.19 sec

The following tests FAILED:
        963 - dft.scan_architect_clock_mix_sky130.tcl (Failed)
        964 - dft.scan_architect_no_mix_sky130.tcl (Failed)
        965 - dft.scan_architect_register_bank_no_clock_mix_sky130.tcl (Failed)
Errors while running CTest
Output from these tests are in: /Users/donn/efabless/openroad/build-mac/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
```

maliberty, 2024-12-23:
With no access to a mac I have no way to debug this.  We make a best effort on mac but make no promises.  Can you track it down?

donn, 2024-12-23:
Will do

PR #6509 (2025-01-18): dft: fix crossplatform inconsistencies
* dft::ScanArchitect: Use a stable sort for scan cells within buckets. Also added a debugPrint to preview the buckets.
  * This is necessary as otherwise two scan elements with identical bit widths and clock domains's order is undefined and a place for inconsistency between platforms.
* dft::ClockDomain::getClockDomainId: Uses the FNV1a hash algorithm which unlike std::hash is not implementation-dependent and will return an identical hash across all platforms.
  * This is necessary as C++ maps sort not by in the insertion order, but by the key order, e.g. [... cut]
closed 2025-01-18 by maliberty