The-OpenROAD-Project/OpenROAD #6431: Error occurs when reading 5.8 def file : "[WARNING ODB-0126] error: MASK on track is unsupported"

jhyun1-shim, 2024-12-26:
### Describe the bug

I'm trying to convert 5.8 version lef/def files to protocol buffer format.
While reading def file, I got this error message below.

OpenROAD v2.0-17585-g87af90f72
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING ODB-0124] warning: Polygon DIEAREA statement not supported.  The bounding box will be used instead
[WARNING ODB-0126] error: MASK on track is unsupported
[WARNING ODB-1003] ERROR (DEFPARS-6010): An error has been reported in callback. Error in file design/ECMP_HWA_TOP.def at line 1411, on token ;.
Update the def file before parsing the file again.

[ERROR ODB-0421] DEF parser returns an error!

The line 1411 in the def file where the error occurred is as follows:
TRACKS Y 256 DO 200 STEP 800 MASK 2 SAMEMASK LAYER M1 ;

According to lefdef reference manual, 'MASK' on track option, which is not included on 5.6 version, is added on 5.8 version.
Doesn't this option support by openroad? or is there any way to read def file that includes this option so that i can solve this problem?
I wonder the latest version of openroad fully support reading 5.8 lef/def files. 

### Expected Behavior

Be able to read DEF 5.8 file fully without any error.
Successful conversion of LEF/DEF to Protocol Buffer format.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.167.4-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-17640-gcc41a1ab9
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 104 lines cut ...]
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP: TRUE (found version "4.5")  
-- GUI is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found suitable version "1.87.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.87.0/BoostConfig.cmake (found version "1.87.0")  
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.Mi3dCOG7jv
```

### To Reproduce

read_def

maliberty, 2024-12-26:
We don't support coloring and tracks.  The most we could do would be to read and ignore that data.

maliberty, 2024-12-26:
The simplest workaround is just to remove the offending data from the DEF.  I am assuming this DEF comes from another tool

PR #6439 (2024-12-28): odb: add mask information to tracks from DEF
Closes #6431
closed 2024-12-28 by maliberty