The-OpenROAD-Project/OpenROAD #6444: Add "power" column to timing report

oharboe, 2024-12-29:
### Description

It would be nice to know the power consumption for a timing path and to see the individual instance powers as a column.

https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6320

### Suggested Solution

Add power column in timing report

gadfort, 2024-12-29:
In the GUI it should be fairly easy to add, for the timing report you will need to request that from OpenSTA. One note of caution, calculating the power is pretty time consuming so it would further slow down the GUI (although since that information is computed separately it might be possible to collect the power numbers on a separate thread)

maliberty, 2024-12-29:
It is costly and you can't call two sta apis from different threads so you would have to pay the full price.  How would it be helpful to see in the timing report?

oharboe, 2024-12-30:
It would help me build a stronger mental model of what draws power, which is useful when working on RTL.

QuantamHD, 2024-12-30:
Why don't we add a button that you can click when you open a timing path that can populate the columns

oharboe, 2024-12-30:
Perhaps this should be a scriptable feature to populate a column with some property of interest to the user.

There is the timing paths table and then the elements within a path, so are you imagining two buttons?

If power is the last property known to man of interest and/or tcl scriptlets cant do anything interesting, then C++ is better.

QuantamHD, 2024-12-30:
I think I'm imagining you open a timing path, and then another button appears that would populate the power of each element in the path.

oharboe, 2024-12-30:
> I think I'm imagining you open a timing path, and then another button appears that would populate the power of each element in the path. 

It would also be useful to have the power for each timing path at the top level before selecting a path.

gadfort, 2024-12-30:
As a side note, the inspector already has the capability to inspect the power information from OpenSTA, which would be a much richer source of information for debugging.

oharboe, 2024-12-30:
> As a side note, the inspector already has the capability to inspect the power information from OpenSTA, which would be a much richer source of information for debugging.

Maybe a right click menu to help the user connect the dots?

"Show in inspector"

?

maliberty, 2024-12-30:
double click already does that

oharboe, 2025-01-07:
> As a side note, the inspector already has the capability to inspect the power information from OpenSTA, which would be a much richer source of information for debugging.

With `make DESIGN_CONFIG=designs/nangate45/gcd/config.mk`.

with `make gui_final`, where do I double click? On one of the lines?

[image]

No power here:

[image]

oharboe, 2025-01-07:
> From the iterm go to the instance

[image]

[image]

maliberty, 2025-01-07:
and there is the "Total power"