The-OpenROAD-Project/OpenROAD #6445: PDN creates ring outside of die area

jeffng-or, 2024-12-30:
### Describe the bug

For some of some of the MegaBoom macros, PDN ends up creating the power ring outside of the die area boundary. As a result, the pins are no longer accessible to the detailed router when we are trying to route BoomTile. The power ring creates an obstruction that doesn't align with the die boundary, which effectively covers the pin shapes:

[image]

In this case of btb_128x56, the ring area is ( -72 144 ) ( 47106 25236 ) and the die area is ( 0 0 ) ( 47060 25232 )

It looks like the shapes are created outside of the die area since that the inner ring core is calculated at ( 240 456 ) ( 46794 24924 ) and then adjusted by the horizontal/vertical width calculation. The net effect is that the ring outline gets grown by 624 (x lo shifted left by 312 and x hi shifted right by 312). Since the inner ring core is starting at 240, the resulting x lo for the ring ends up at -72.

### Expected Behavior

Either the ring is created within the die boundary or we error out if it isn't possible.

### Environment

```shell
My local workspace is pointing to:

commit ad1b24dbd83dbaadaf5b5df8effc6c9d643b1421 (HEAD -> master, origin/master, origin/HEAD)
Merge: 0812b358 261a6266
Author: Matt Liberty <mliberty@precisioninno.com>
Date:   Mon Dec 23 15:15:32 2024 -0800

    Merge pull request #2620 from The-OpenROAD-Project-staging/secure-yosys0.48
    
    update Yosys to version 0.48
```

### To Reproduce

Test case can be found at

https://drive.google.com/file/d/10UJD8IatCit-IcZ3RZsMG0kk4vgCjEhh/view?usp=drive_link

maliberty, 2024-12-30:
@gadfort are you available to look at this?

gadfort, 2024-12-30:
@maliberty I believe this was intentional because openlane was inserting rings outside of the die area and then adjusting their die area. Personally I agree that the tools should not generate shapes outside of the die area and I'm happy to change the behavior to error out (it's easy to detect so should not require much to implement). If we are okay with changing this behavior to be an error, I can do that.

maliberty, 2024-12-30:
It feels backwards to get a ring outside and then post-adjust the die area.  @donn do you have any concerns about making this an error?

@jeffng-or are you able to modify the pdn setup to avoid this issue?

donn, 2024-12-30:
I _believe_ this was before my time, so I'd have to figure out why with the team first. I'll get back to you.

jeffng-or, 2024-12-30:
The SRAM flow is using the [BLOCK_grid_strategy.tcl](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/openRoad/pdn/BLOCK_grid_strategy.tcl) directly from the asap7 platforms area, so I'm not sure how much flexibility I have to modify it.

The SRAMs are all generated using the mock area flow in megaboom, so I could look at what needs to be tweaked in order to avoid the issue. Of the fifteen SRAM types in megaboom, nine have this obstruction issue.

@oharboe FYI.

oharboe, 2024-12-31:
A surprising behavior....

https://en.wikipedia.org/wiki/Principle_of_least_astonishment

maliberty, 2024-12-31:
The problem here is the floorplan setup isn't compatible with the pdn setup.  In pdn
```
add_pdn_ring    -grid {top} -layers {M5 M4} -widths {0.12 0.12} -spacings {0.072} -core_offset {0.084}
```
which means you will need at least .084 + 12 + .072 + .12 = .396 between core and die edge to fit the rings inside.  But you only have 0.324
[image]
How did you determine the core spacing?

maliberty, 2024-12-31:
Most designs in ORFS are using at least 1 micron and so don't have this issue:
```
aes-block/config.mk:13:export CORE_MARGIN            = 2
aes-mbff/config.mk:13:export CORE_MARGIN              = 2
aes/config.mk:13:export CORE_MARGIN              = 2
aes_lvt/config.mk:13:export CORE_MARGIN            = 2
ethmac/config.mk:11:export CORE_MARGIN            = 2
ethmac_lvt/config.mk:12:export CORE_MARGIN            = 2
gcd/config.mk:9:export CORE_AREA              = 1.08 1.08 15.12 15.12
ibex/config.mk:11:export CORE_MARGIN            = 2
jpeg/config.mk:13:export CORE_MARGIN            = 2
jpeg_lvt/config.mk:22:export CORE_MARGIN            = 2
mock-cpu/config.mk:11:export CORE_MARGIN              = 2
riscv32i/config.mk:23:export CORE_AREA = 5 5 75 85 
swerv_wrapper/config.mk:21:export CORE_AREA   = 5 5 545 595 
uart/config.mk:11:export CORE_AREA              = 1.08 1.08 16 16
```

oharboe, 2024-12-31:
@louiic Made the BLOCK_ and BLOCKS_ .tcl files for mock-array

Later, I have been using them in test designs where there are macros built by OpenROAD and a top level.

maliberty, 2024-12-31:
There is nothing wrong with the BLOCK* files - you have to use them intelligently.  If you lower the core margin too much you get this problem.

oharboe, 2024-12-31:
An actionable error message would be instructive.

maliberty, 2024-12-31:
That is pending on Donn's response as to whether they need this behavior.

donn, 2025-01-01:
So, we don't know what you mean by the resizing thing, probably a one-off thing in the early days, but what we *do* know is that we intentionally create PDN rings outside the die area regardless.

This is the XOR template for the power grid for caravel user project: the I/O pins are at the edge of the die area and the rings surround the I/Os. 

[image]

Should PDNs outside the die area be disallowed, we would have to update a ton of layout files and configuration files so that the PDN goes inside the die area, and the pins are not laid out across the die area boundaries. Additionally, this would allow routes for the pins to be created in the new area, which we do not want to happen, so we'd have to add blockages… The drawbacks keep piling up. 

We do understand your concern however, as this IS surprising behavior for 99% of users. We would be okay with updating the default behavior, but with an escape hatch (`-allow_outside_die_area`) to revert to the previous behavior.

gadfort, 2025-01-01:
A flag seems like a reasonable idea to me.

maliberty, 2025-01-01:
Perhaps it would be better to compute the bounds per layer in write_lef_abstract and leave this as a warning?

maliberty, 2025-01-13:
@donn ping

donn, 2025-01-14:
@kareefardi ?

kareefardi, 2025-01-19:
[image]

[image]

We use magic to generate the abstract LEF. Magic doesn't draw obs over pins. Instead, it uses min and max coordinates of metal shapes that don't overlap pins and creates wrapping obstructions around the pins. Since there is no metal wire outside the PR boundary, the resultant LEF allows design pin access as shown in the images

maliberty, 2025-01-19:
That's interesting.  We could change the LEF abstract generator to be similar.  It is unnecessary to cut the obs around the pins though.  From the manual

```
Normally, obstructions block routing, except for when a pin port overlaps an obstruction (a
port geometry overrules an obstruction). For example, you can define a large rectangle for a
metal1 obstruction and have metal1 port in the middle of the obstruction. The port can still
be accessed by a via, if the via is entirely inside the port. 
```

maliberty, 2025-01-22:
After discussing with @gadfort it looks like the best is to make it an error with an option to allow for eFabless' use case.

PR #6583 (2025-01-23): pdn: no outside die shapes
Closes: #6445

Changes:
- adds flag to permit pdn rings outside the die, changes the default to not allowing it
- Updates tests that had rings outside the die boundaries.
closed 2025-01-23 by maliberty