The-OpenROAD-Project/OpenROAD #6451: Not enough parameters passed to `SWIG_Python_AppendOutput()` with swig 4.3

hzeller, 2024-12-31:
### Description

When compiling with swig 4.3, the generated `SWIG_Python_AppendOutput()` function gets an additional parameter `int is_void` which then results in compile error when called with the original two parameters.

I noticed that on a system compiling with swig 4.3; maybe this started earlier, but can't verify right now. Someone with more knowledge about swig probably can pinpoint the situation better.

### Suggested Solution

In a local hack I could make OpenROAD compile by adding a `0 `parameter in dbtypes.i:

```patch
--- a/src/odb/src/swig/python/dbtypes.i
+++ b/src/odb/src/swig/python/dbtypes.i
@@ -271,7 +271,7 @@ WRAP_OBJECT_RETURN_REF(odb::dbViaParams, params_return)
   swig_type_info *tf = SWIG_TypeQuery("odb::dbShape" "*");
   for(std::vector<odb::dbShape>::iterator it = $1->begin(); it != $1->end(); it++) {
     PyObject *o = SWIG_NewInstanceObj(&(*it), tf, 0);
-    $result = SWIG_Python_AppendOutput($result, o);
+    $result = SWIG_Python_AppendOutput($result, o, 0);
   }
 }
 
@@ -283,14 +283,14 @@ WRAP_OBJECT_RETURN_REF(odb::dbViaParams, params_return)
     auto layer = it->second;
     PyObject *layer_swig = SWIG_NewInstanceObj(layer, tf, 0);
     PyObject *tuple = PyTuple_Pack(2, PyFloat_FromDouble(value), layer_swig);
-    $result = SWIG_Python_AppendOutput($result, tuple);
+    $result = SWIG_Python_AppendOutput($result, tuple, 0);
   }
 }
 
 %typemap(argout) std::vector<int> &OUTPUT {
   for(auto it = $1->begin(); it != $1->end(); it++) {
     PyObject *obj = PyInt_FromLong((long)*it);
-    $result = SWIG_Python_AppendOutput($result, obj);
+    $result = SWIG_Python_AppendOutput($result, obj, 0);
   }
 }
```

This is of course a local hack that should probably done with some `ifdef` of sorts or other version-detecting technique

gadfort, 2024-12-31:
See: https://github.com/The-OpenROAD-Project/OpenROAD/issues/6139

hzeller, 2025-01-03:
According to https://www.swig.org/Release/CHANGES.current it is probably good to replace `SWIG_Python_AppendOutput()` with `SWIG_AppendOutput()`

```
2024-06-15: vadz
            [Python] #2907 Fix returning null from functions with output
            parameters.  Ensures OUTPUT and INOUT typemaps are handled
            consistently wrt return type.

            New declaration of SWIG_Python_AppendOutput is now:

              SWIG_Python_AppendOutput(PyObject* result, PyObject* obj, int is_void);

            The 3rd parameter is new and the new $isvoid special variable
            should be passed to it, indicating whether or not the wrapped
            function returns void. If SWIG_Python_AppendOutput is currently being
            used and a completely backwards compatible (but technically incorrect)
            solution is required, then pass 1 for the is_void parameter.

            Also consider replacing with:

              SWIG_AppendOutput(PyObject* result, PyObject* obj);

            which calls SWIG_Python_AppendOutput with same parameters but adding $isvoid
            for final parameter.

	    *** POTENTIAL INCOMPATIBILITY ***
```

maliberty, 2025-01-03:
@hzeller do you want to make a PR with that change?