The-OpenROAD-Project/OpenROAD #6453: cmake configure fails for sta tests recently

hzeller, 2024-12-31:
### Description

After commit 3e9fa0816263d0e1356f251f4a09840f98f8cd4c I get an error when configuring (with `-DUSE_SYSTEM_OPENSTA=OFF`)

```
CMake Error at src/dbSta/test/cpp/CMakeLists.txt:19 (gtest_discover_tests):
  Unknown CMake command "gtest_discover_tests".
```

When adding `include(GoogleTest)` to `src/dbSta/test/cpp/CMakeLists.txt` to address the `gtest_discover_tests` not found issue, there is another one:

```
CMake Error at src/dbSta/test/cpp/CMakeLists.txt:24 (add_dependencies):
  Cannot add target-level dependencies to non-existent target
  "build_and_test".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.
```

So it looks like something needs to be wired up differently in CMakeLists for this to work

### Suggested Solution

Don't know enough `cmake` to know how to fix this.

### Additional Context

@andyfox-rushc might know what to do.

maliberty, 2025-01-01:
I tried it on ubuntu22 and got no error.  What version of cmake and OS are you using?

hzeller, 2025-01-01:
Cmake version 3.30.5 on nixos.

Also, one possibly relevant thing I forgot above: it was with `-DENABLE_TESTS=OFF` passed to the toplevel cmake.

I am packaging OpenROAD for nix and had to [skip the directory](https://github.com/NixOS/nixpkgs/blob/2205bbfd7ef891ddcfa251df9d2a59b2c0f00528/pkgs/applications/science/electronics/openroad/skip-sta-gtest.patch) to make it compile.

hzeller, 2025-01-01:
It is probably just a matter of

```patch
--- a/src/dbSta/test/CMakeLists.txt
+++ b/src/dbSta/test/CMakeLists.txt
@@ -51,6 +51,6 @@ or_integration_tests(
 foreach(TEST_NAME IN LISTS TEST_NAMES)
     or_integration_test("dbSta" ${TEST_NAME} ${CMAKE_CURRENT_SOURCE_DIR}/regression)
 endforeach()
-
-add_subdirectory(cpp)
-
+if(ENABLE_TESTS)
+  add_subdirectory(cpp)
+endif
```

will send a pull request.

PR #6455 (2025-01-01): sta: only add test subdirectory if tests enabled.
Fixes: #6453
commit 77defcdd (2025-01-01): sta: only add test subdirectory if tests enabled.

Fixes: #6453
closed 2025-01-01 by eder-matheus