The-OpenROAD-Project/OpenROAD #6456: Strange power numbres with ASAP7

oharboe, 2025-01-01:
OpenROAD v2.0-17847-g43e602d7b

https://drive.google.com/file/d/1bq3ULEK2iD4Q0-ilI1IE4K_QX9qfVse3/view?usp=sharing

to inspect results untar and run:

```
. vars-fs-mul-mod-q-asap7-base.sh
ODB_FILE=results/asap7/fs-mul-mod-q/base/6_final.odb openroad -gui flow/scripts/open.tcl
```

[image]

```
>>> report_power
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.16e-03   7.67e-06   3.04e-08   1.17e-03   0.3%
Combinational          2.06e-01   1.95e-01   2.05e-06   4.01e-01  99.7%
Clock                  8.88e-05   4.72e-05   9.05e-09   1.36e-04   0.0%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.07e-01   1.95e-01   2.09e-06   4.02e-01 100.0%
                          51.5%      48.5%       0.0%
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6320#discussioncomment-11689607_

oharboe, 2025-01-01:
@jjcherry56 @maliberty @evmanz Standard format bug report.

oharboe, 2025-01-04:
Tried with https://github.com/parallaxsw/OpenSTA/commit/87418f23b1f5c02b0e3334023aedabfefd9b8025

```
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.16e-03   7.67e-06   3.04e-08   1.17e-03   0.3%
Combinational          2.06e-01   1.95e-01   2.05e-06   4.01e-01  99.7%
Clock                  8.88e-05   4.72e-05   9.05e-09   1.36e-04   0.0%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.07e-01   1.95e-01   2.09e-06   4.02e-01 100.0%
                          51.5%      48.5%       0.0%
```

oharboe, 2025-01-05:
@maliberty Unchanged after https://github.com/parallaxsw/OpenSTA/commit/87418f23b1f5c02b0e3334023aedabfefd9b8025

tspyrou, 2025-01-06:
From discussion https://github.com/The-OpenROAD-Project/OpenROAD/discussions/6320

jjcherry56, 2025-01-09:
I gave up trying to run this because there is so much garbage like this
set env(BC_CCS_DFF_LIB_FILE) "$::env(FLOW_HOME)/../foo/test-case/../../flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_FF_ccs_220123.lib"

in
vars-fs-mul-mod-q-asap7-base.tcl

evmanz, 2025-01-10:
[Here is the archive](https://drive.google.com/file/d/1QusX_IPB3Z4n1PF4HzCQx-Gbq9c6FT9n/view?usp=share_link) made with ```make detail_route_issue```

oharboe, 2025-01-10:
> I gave up trying to run this because there is so much garbage like this set env(BC_CCS_DFF_LIB_FILE) "$::env(FLOW_HOME)/../foo/test-case/../../flow/platforms/asap7/lib/NLDM/asap7sc7p5t_SEQ_RVT_FF_ccs_220123.lib"
> 
> in vars-fs-mul-mod-q-asap7-base.tcl

Sorry about that, `make issue` generated broken vars-fs-mul-mod-q-asap7-base.* files and I only fixed the .sh variant.

I never use the .gdb or .tcl versions, only the .sh version.

I would give the `vars-*.tcl/.db` files from `make issue` a wide berth and I've suggested that it should be retired as it obviously hasn't been tested in quite some time https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2670

@maliberty FYI

jjcherry56, 2025-01-10:
vars_mumble.tcl is the only useful file as far as I am concerned.
I debug inside emacs by sourcing vars.tcl in .openroad so shell scripts are worthless.
Currently I have to set env(FLOW_HOME) and FLOW_HOME for the broken vars.tcl file to work.

maliberty, 2025-01-29:
still relevant?

evmanz, 2025-01-29:
> still relevant?

yep. 

I tried the latest last week, and the power situation is the same.

tspyrou, 2025-01-30:
@evmanz were you able to run make issue so that it will be straightforward for us to reproduce what you are doing?
We would like to look at this.

evmanz, 2025-01-31:
> [@evmanz](https://github.com/evmanz) Did I transport the problem correctly into this github issue? [parallaxsw/OpenSTA#201](https://github.com/parallaxsw/OpenSTA/issues/201)

yep! 

I am a bit confused. I am using [OpenROAD-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts), and there is no `make issue` in OpenROAD-flow-scripts/flow/Makefile. Is the `issue` target in a different location? 

@tspyrou, @oharboe did it.

maliberty, 2025-01-31:
util/utils.mk

evmanz, 2025-01-31:
> util/utils.mk

Well, it has %_issue, which means it's an issue per stage, right? Like, make cts_issue. Correct?

maliberty, 2025-01-31:
Yes, you can package an issue from any step

oharboe, 2026-01-06:
this is fixed.

However, we use read_vcd/saif to avoid mysteries altogether about how pin wiggling is propagated