The-OpenROAD-Project/OpenROAD #6457: Reduce click and wait in the GUI by saving heatmap calculations in .odb

oharboe, 2025-01-02:
### Description

Calculating heatmaps in the GUI can take a long time for big designs.

Some experimentation to move this calcuation prior to opending the GUI was done[Some experimentation](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2610).

However, it was concluded that what is really needed is to save such caclulations in the .odb file for the best user experience.

### Suggested Solution

Save heatmap calculations done during the build in the .odb file to reduce click+wait in the GUI in the final step. I prefer default on: this is fast for tiny designs and big designs take long anyway, so might as well reduce click+wait in the GUI.

gadfort, 2025-01-02:
@oharboe could you provide some details about what heatmaps take a long time? What size bins are you using?
The way the heatmaps are configured it doesn't lend itself to saving the data to the database.

oharboe, 2025-01-02:
Test case created from 

```
$ bazel run BoomTile_grt $(pwd)/grt
$ grt/make open_issue
```

untar and run https://drive.google.com/file/d/1YOnr4QQg4e2gvvHcFpfNB0JxgatEabS3/view?usp=sharing

```
$ ODB_FILE=results/asap7/BoomTile/base/5_1_grt.odb ./run-me-BoomTile-asap7-base.sh
OpenROAD v2.0-17943-g06a891e4a 
Features included (+) or not (-): +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
GUI_TIMING=1 reading timing, takes a little while for large designs...
No global routing results available, skipping estimate_parasitics
Load .//reports/asap7/BoomTile/base/congestion.rpt for details
Populating timing paths...OK
openroad> gui::show
```

30 seconds in this case, cases I can't share can take longer:

[image]