The-OpenROAD-Project/OpenROAD #6461: make openroad able to run power.tcl in mock-array

oharboe, 2025-01-02:
### Description

untar and run https://drive.google.com/file/d/1y1PP7ZeJnzEm9BO-X5oYXjpNZO4EggWF/view?usp=sharing

power.tcl works when invoked with `sta -no_init -exit power.tcl`, but fails with openroad, which is surprising.

```
[deleted]
+ openroad -no_init -exit power.tcl
OpenROAD v2.0-17943-g06a891e4a 
Features included (+) or not (-): +GPU +GUI +Python
[deleted]
[ERROR ORD-2010] no technology has been read.
Error: power.tcl, 11 ORD-2010
```

### Suggested Solution

Make it work with `openroad` as well as `sta`

maliberty, 2025-01-02:
OR requires LEF and sta does not.  Just load the LEF.

oharboe, 2025-01-03:
I tried https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2651, but got:

```
link_design MockArray
[WARNING STA-0349] instance 'ces_0_0' not found.
[WARNING STA-0349] instance 'ces_0_1' not found.
[WARNING STA-0349] instance 'ces_0_2' not found.
[WARNING STA-0349] instance 'ces_0_3' not found.
[WARNING STA-0349] instance 'ces_0_4' not found.
[WARNING STA-0349] instance 'ces_0_5' not found.
[WARNING STA-0349] instance 'ces_0_6' not found.
[WARNING STA-0349] instance 'ces_0_7' not found.
[WARNING STA-0349] instance 'ces_1_0' not found.
[deleted]
```

oharboe, 2025-01-09:
out of sight, out of mind for now.

closed 2025-01-09 by oharboe