The-OpenROAD-Project/OpenROAD #6490: build: cmake gets stuck when building

titan73, 2025-01-08:
### Describe the bug

This comes from the following commit:
```
commit 5603ac23198d1266b5e8e89ef68af4d482b9dd3a
Author: Ethan Mahintorabi <ethanmoon@google.com>
Date:   Tue Jan 7 20:38:18 2025 +0000

    removing stray semi-colon
    
    Signed-off-by: Ethan Mahintorabi <ethanmoon@google.com>
```
It's not specified but this commits also changes abc from ef5389d31526003c2ebd7e6d6d6fe3848a20f0a2 to 5c9448c085eb8bf1e433a22a954532e44206c6f9. Is it intended?

It seems this is abc change that causes the issue.

### Expected Behavior

Should not get stuck

### Environment

```shell
Lastest OR
```

### To Reproduce

cd build
cmake ..

### Relevant log output

```shell
-- OpenROAD version: v2.0-18036-g210dc45e9
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17
-- Performing Test C_COMPILER_SUPPORTS__-std=c++17 - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17
-- Performing Test CXX_COMPILER_SUPPORTS__-std=c++17 - Success
```

titan73, 2025-01-08:
Bisecting abc shows the commit:
```
commit 034492b0e250ee67f053f6c59ad2a3a7cf34a0eb
Merge: af77b8019 49a489554
Author: alanminko <37236958+alanminko@users.noreply.github.com>
Date:   Sat Aug 24 09:21:50 2024 -0700

    Merge pull request #322 from rocallahan/unit-tests
    
    Adds unit testing framework to ABC
```
which merges:
```
commit 034492b0e250ee67f053f6c59ad2a3a7cf34a0eb
Merge: af77b8019 49a489554
Author: alanminko <37236958+alanminko@users.noreply.github.com>
Date:   Sat Aug 24 09:21:50 2024 -0700

    Merge pull request #322 from rocallahan/unit-tests
    
    Adds unit testing framework to ABC
```
Indeed reverting 034492b0e250ee67f053f6c59ad2a3a7cf34a0eb fixed the issue.

titan73, 2025-01-08:
Il let it run for a while an finally I got an error:
```
CMake Error at googletest-subbuild/googletest-populate-prefix/src/googletest-populate-stamp/download-googletest-populate.cmake:170 (message):
  Each download failed!

    error: downloading 'https://github.com/google/googletest/archive/refs/tags/v1.14.0.zip' failed
          status_code: 6
          status_string: "Couldn't resolve host name"
          log:
          --- LOG BEGIN ---
          getaddrinfo(3) failed for github.com:443

  Could not resolve host: github.com

  Closing connection 0
```
This is probably similar to https://github.com/The-OpenROAD-Project/OpenROAD/issues/5530

titan73, 2025-01-08:
cmake .. -DABC_SKIP_TESTS="" workarounds the issue.

maliberty, 2025-01-08:
"Couldn't resolve host name" looks like a local networking issue.  I don't see an OR fix here.

closed 2025-01-08 by maliberty

titan73, 2025-01-08:
Yes indeed it's in abc. The issue is due to the firewall as for the mentionned fixed issue.

maliberty, 2025-01-08:
What is requested here?  You have a workaround to your local issue and I don't see something more general to do.

titan73, 2025-01-08:
A similar thing could be done in abc but abc does not have a script to install dependencies and gtest versions between abc and OR does not match anyway (1.4.0 vs 1.3.0). Not worth the trouble.
So nothing to do actually, the workaround is good for me as I don't intend to run abc tests. This issue can stay closed. Thanks